mit-han-lab/bevfusion #394: More info on meta_keys in case of custom sensors setup

VeeranjaneyuluToka, 2023-04-21:
Hi @kentang-mit  and all, I have custom sensor setup to generate in house dataset, camera and LiDAR are placed at same position which is defined as ego position as of now. So the meta_keys would be as follows, is not it?

**camera2ego**: should be an identity matrix as camera placed at ego position
**lidar2ego**: should also be an identity matrix as LiDAR also be placed at ego position
**lidar2camera**: does this mean LiDAR coordinate system to Camera coordinate system transformation? for example assume if LiDAR is right handed and camera is in left handed, then we need to provide transformation matrix which converts from right handed to left handed?
**camera2lidar**: similar to above
**lidar2image**: this is to convert from 3D lidar coordinates to 2D image coordinates, for example lidar coordinates (Lx, Ly, Lz) -> camera coordinates (Cx, Cy, Cz) -> img coordinates (Ix, Iy)

I could see that these meta-keys are being used in **forward** function but wondering is there any where else are these being used? and how GT bbox gets transformed and where?

Would be great help if somebody can through some glance on this, thanks!

kentang-mit, 2023-05-02:
Hi @VeeranjaneyuluToka,

Sorry for the delayed response. I was busy working on other projects recently. I think these are all fields that will be used in LSS. I would say that `camera2ego` and `lidar2ego` might not be that important after we update our code back in September last year. These two fields were originally used to get `lidar2camera` and `cmaera2lidar`, and it should be good if you use identity matrices for them. 

Best,
Haotian

VeeranjaneyuluToka, 2023-05-03:
@kentang-mit , thanks for your reply! I have a few more questions still on this.

Looks like Lidar2Camera and Camera2Lidar contains extrinsics of sensors, but how the the coordinate transformation happens from (East Down Forward to Forward Left Up) etc.. ?

One more thing is that, Nuscenes lidar point clouds are in ENU (East North Up) but looks like in coversion fuctions, that conversion is not that (LiDAR seems to be in FLU (Forward Left Up), would you comment on this?

Thanks!

kentang-mit, 2023-05-22:
Hi @VeeranjaneyuluToka,

I just came back from another conference deadline. The projection from camera coordinate system to LiDAR coordinate system is done using the two extrinsics matrices you mentioned (i.e. lidar2camera and camera2lidar). For example, `lidar2camera @ lidar_coords` should be `camera_coords`. We also have the image coordinate system (i.e. coordinates in the pixel array when you store the image). This can be obtained using `camera_intriscs @ camera_coords` (the code you listed). For the nuScenes point cloud coordinates, I think you are right, they are in ENU. I'm not sure whether you are referring to the [LiDARInstance3DBoxes](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/core/bbox/structures/lidar_box3d.py#L10) coordinate system when mentioning "LiDAR seems to be in FLU". If it is the case, I guess it does not really matter, since the box parameters are only used to calculate IoU etc. for target assignment during training.

Regarding the coordinate system of all bounding boxes when training the camera-only model, we actually do not convert them to the camera coordinate system. Instead, we project all pixels from the camera space to the LiDAR coordinate system using [this function](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/vtransforms/base.py#L79). So all the supervision is done in the LiDAR coordinate system even for camera-only models. 

Best,
Haotian

closed 2023-07-14 by zhijian-liu