mit-han-lab/bevfusion #396: TensorRT support

lucianzhong, 2023-04-24:
Thanks a lot for this great work on multi-modal perception.

Just wondering how's going with the TensorRT support ?

Best regards

kentang-mit, 2023-05-02:
Hi @lucianzhong,

Sorry about the delayed response. We have limited bandwidth and did not have further explorations on this direction. By the way, because of the fact that sparse convolution layers (used in the LiDAR branch) are hard to deploy with TensorRT, we are actually also considering using AITemplate to deploy the 2D layers and run sparse convolution with TorchSparse, which is a high-performance GPU library for sparse 3D convolution led by me (a new version is coming, but we are still cleanning up the code for release and working on wheels generation).

Best,
Haotian

lucianzhong, 2023-05-03:
Thanks for the reply. Looking forward to the coming new version. 
Best regards

sandeepnmenon, 2023-06-02:
This is great! Looking forward to this.
I see the recurrent issue of sparse convolutions difficult to deploy with TensorRT. Is there a workaround for this?
Almost all 3D lidar models now use sparse convolutions (torchsparse is amazing!)
If I port the entire model to torchsparse, would that be comparable to TensorRT runs?

closed 2023-07-14 by zhijian-liu