mit-han-lab/bevfusion #397: Data preparation

SMSajadi99, 2023-04-25:
What is the purpose of preparing mmdetection3d folder?  

Because these folders were not used at all in the evaluation and training section? 
 
My question is, why is it not used, especially in education?

SMSajadi99, 2023-04-25:
I have now created the folders based on the Github that you said, but when I went to the evaluation section or even the training in the codes, I realized that it has nothing to do with the mmdetection3d folder and the generated pkls.  Why ?

lixinghe1999, 2023-04-28:
pkls seems required, but the mmdetection3d folder is just a optional folder name

SMSajadi99, 2023-04-28:
Hi @lixinghe1999 
I had two questions:
1- Where exactly should these pkl files be placed, because now I made these files based on GitHub mmdetection3d, but I don't know where to put them in bevfusion. (I downloaded your gates, ran the setup file, downloaded the Pre-Train, but where should I put this mmdetection3d and pkl files)

bevfusion folder:

[image]

mmdetection3d folder:

[image]

4 PLK's files:

[image]

2- Of course, I asked in the previous question, but where should I put this mmdet3d or mmdetection3d folder from the mmdetection3d GitHub?

SMSajadi99, 2023-04-28:
Because I looked at your github, I didn't understand exactly where to put the mmdet3d (or mmdetection3d) folder and the PLK files.

lixinghe1999, 2023-04-28:
1. pkls will go to the same folder of "maps", "samples" and "sweeps"
2. from my understanding, this repo is forked from mmdet3d, no need to download another mmdet3d again, also this repo doesn't require "pip install mmdet3d"

SMSajadi99, 2023-04-28:
Maybe I didn't say my opinion correctly.
Now, according to the bevfusion GitHub, we should have the following folders and prepare it in this way

[image]

Now my question is, where should we put this `mmdetection3d `folder? Should we work with this folder at all?

lixinghe1999, 2023-04-28:
from my experience, this figure is a little bit misleading.
"mmdetection3d" == "bevfusion"(just the name of this repo),
what you need to do is just clone this repo, mkdir data, download dataset into ./data, and add pkls.

SMSajadi99, 2023-04-28:
One more question, in the bevfusion folder, should we remove the data folder from the mmdetection3d github and put it there?

Is this folder enough? No other folder is needed?

SMSajadi99, 2023-04-28:
Hi @zhijian-liu and @kentang-mit 

Thank you very much for the project and the good work you are doing.
I may be sorry for your trouble, if you please give me some guidance, I will be grateful.

kentang-mit, 2023-05-02:
Hi @SMSajadi99,

Sorry about the delayed response. I was engaged in other projects recently and I finally got some time to look at bevfusion issues this week. I think most of the explanations of @lixinghe1999 are helpful. As he said, `mmdet3d` is just a folder name and I would suggest you to forget about the latest mmdet3d repo when doing experiments using our codebase because we actually cloned it around 1.5-2 years ago and made a lot of changes on top of it. You could clone this repo, view it as a new one and first run 

```bash
python setup.py develop
```

to install the repo locally. Afterwards, regarding the structure of the `data` folder, what I usually do is to symbolic link `data/nuscenes` to the actual location where the nuScenes dataset and prepared PKL files are stored. The current structure of this `data/nuscenes` folder you show looks correct to me.

Best,
Haotian

SMSajadi99, 2023-05-13:
Hi @kentang-mit && @lixinghe1999 
Thank you very much for the advice.

I did this, I placed the `pkl`s and data, but I encountered a problem in the evaluation, can you help me?

`(Sajadi_BEVFusion_env) [sajadi@edahost bevfusion]$ torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
/run/media/msharifk/Data/anaconda3/envs/Sajadi_BEVFusion_env/bevfusion/mmdet3d/core/bbox/box_np_ops.py:781: NumbaDeprecationWarning: The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.                               
  def points_in_convex_polygon_jit(points, polygon, clockwise=True):                                
Config (path: configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml): {'seed': 0, 'deterministic': False, 'checkpoint_config': {'interval': 1, 'max_keep_ckpts': 1}, 'log_config': {'interval': 50, 'hooks': [{'type': 'TextLoggerHook'}, {'type': 'TensorboardLoggerHook'}]}, 'load_from': None, 'resume_from': None, 'cudnn_benchmark': False, 'fp16': {'loss_scale': {'growth_interval': 2000}}, 'max_epochs': 6, 'runner': {'type': 'CustomEpochBasedRunner', 'max_epochs': 6}, 'dataset_type': 'NuScenesDataset', 'dataset_root': 'data/nuscenes/', 'gt_paste_stop_epoch': -1, 'reduce_beams': 32, 'load_dim': 5, 'use_dim': 5, 'load_augmented': None, 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0], 'voxel_size': [0.075, 0.075, 0.2], 'image_size': [256, 704], 'augment2d': {'resize': [[0.38, 0.55], [0.48, 0.48]], 'rotate': [-5.4, 5.4], 'gridmask': {'prob': 0.0, 'fixed_prob': True}}, 'augment3d': {'scale': [0.9, 1.1], 'rotate': [-0.78539816, 0.78539816], 'translate': 0.5}, 'object_classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'map_classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider'], 'input_modality': {'use_lidar': True, 'use_camera': True, 'use_radar': False, 'use_map': False, 'use_external': False}, 'train_pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ObjectPaste', 'stop_epoch': -1, 'db_sampler': {'dataset_root': 'data/nuscenes/', 'info_path': 'data/nuscenes/nuscenes_dbinfos_train.pkl', 'rate': 1.0, 'prepare': {'filter_by_difficulty': [-1], 'filter_by_min_points': {'car': 5, 'truck': 5, 'bus': 5, 'trailer': 5, 'construction_vehicle': 5, 'traffic_cone': 5, 'barrier': 5, 'motorcycle': 5, 'bicycle': 5, 'pedestrian': 5}}, 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'sample_groups': {'car': 2, 'truck': 3, 'construction_vehicle': 7, 'bus': 4, 'trailer': 6, 'barrier': 2, 'motorcycle': 6, 'bicycle': 6, 'pedestrian': 2, 'traffic_cone': 2}, 'points_loader': {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32}}}, {'type': 'ImageAug3D', 'final_dim': [256, 704], 'resize_lim': [0.38, 0.55], 'bot_pct_lim': [0.0, 0.0], 'rot_lim': [-5.4, 5.4], 'rand_flip': True, 'is_train': True}, {'type': 'GlobalRotScaleTrans', 'resize_lim': [0.9, 1.1], 'rot_lim': [-0.78539816, 0.78539816], 'trans_lim': 0.5, 'is_train': True}, {'type': 'LoadBEVSegmentation', 'dataset_root': 'data/nuscenes/', 'xbound': [-50.0, 50.0, 0.5], 'ybound': [-50.0, 50.0, 0.5], 'classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider']}, {'type': 'RandomFlip3D'}, {'type': 'PointsRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectNameFilter', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'ImageNormalize', 'mean': [0.485, 0.456, 0.406], 'std': [0.229, 0.224, 0.225]}, {'type': 'GridMask', 'use_h': True, 'use_w': True, 'max_epoch': 6, 'rotate': 1, 'offset': False, 'ratio': 0.5, 'mode': 1, 'prob': 0.0, 'fixed_prob': True}, {'type': 'PointShuffle'}, {'type': 'DefaultFormatBundle3D', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'Collect3D', 'keys': ['img', 'points', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_masks_bev'], 'meta_keys': ['camera_intrinsics', 'camera2ego', 'lidar2ego', 'lidar2camera', 'camera2lidar', 'lidar2image', 'img_aug_matrix', 'lidar_aug_matrix']}], 'test_pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps' [... cut ...]
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[18480,1],1]
  Exit code:    1
--------------------------------------------------------------------------
`

SMSajadi99, 2023-05-13:
Hi again @kentang-mit && @lixinghe1999

Now the problem is that it does not understand the word `info` in the last line of the above message, why? What should I do to solve this problem?
Because just before this error it gave an error that we need to install `numba` and I installed version `0.57` which is the latest version, of course, as you said on the Github page, the `Python` version should be `3.8`.

SMSajadi99, 2023-05-13:
https://numba.pydata.org/numba-doc/latest/user/installing.html
https://pypi.org/project/numba/
I cheked it for installing `numba`

SMSajadi99, 2023-05-18:
Hi @kentang-mit 
I made some changes to the code to work with mini data.
The following photo is my changes:

[image]

I tried to change its version to mini, and after that, we don't have `infos `in the data keys, it only had `data_list `and `metainfo`, which I changed in the circled part so that it can understand it, in addition to that, `gcc `I upgraded it as if above 5 was intended, the `numba `library was also necessary, I installed it, then I typed the following code for evaluation: (from this link https://github.com/mit-han-lab/bevfusion#evaluation)

`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`

After downloading a cache for Torch still remains in this section:

[image]

I got the `pkl`s from this link that I asked one of your friends and I am working with them

https://github.com/open-mmlab/mmdetection3d/issues/2454#issuecomment-1517743436

What needs to be done?

SMSajadi99, 2023-05-19:
Based on the question I asked, the problem was solved:
https://github.com/mit-han-lab/bevfusion/issues/409

But I ran into another problem:
[image]

Now I checked the code, there is a problem in this part:
[image]

As you can see, we don't have `lidar_path `or `location `or `sweeps `at all.

what should I do ?

SMSajadi99, 2023-07-12:
Hello again, @kentang-mit 
I tried very hard to optimize the codes for the `mini` version, but I couldn't. Do you have any practical instructions so that I can do this easier

closed 2024-05-04 by zhijian-liu