mit-han-lab/bevfusion #398: Pretrain Model

liyiheng0913, 2023-04-25:
Hello，I can't download the pretrain model in tools/download_pretrained.sh，is there any other way to download the pretrain model？

qianqianqianqian, 2023-04-26:
Same question, could you provide bevfusion-L pre-trained model, I reproduce for a long time but get 0.6nds lower than reported.
And how about scaling up the MACs and image size experience, I try it but OOM(on 80g A100),  is there some solution?
Thanks a lot!

atlantins, 2023-04-26:
Same question

iyaqiao, 2023-04-28:
Same question,I also can't download the pretrain model !

kentang-mit, 2023-05-02:
Hi folks, I mentioned [elsewhere](https://github.com/mit-han-lab/bevfusion/issues/305) that there is a small bug in the center of GT boxes. I ran some experiments after fixing this and I was able to get the reported results quite often.

kentang-mit, 2023-05-02:
Regarding pretrained models, we noticed that our lab website was not accessible in some regions. You could try out VPNs to see whether it is possible to download these checkpoints. If it is still difficult, please send me an Email and I will try to send a copy to Google Drive.

Mandylove1993, 2023-05-06:
> Regarding pretrained models, we noticed that our lab website was not accessible in some regions. You could try out VPNs to see whether it is possible to download these checkpoints. If it is still difficult, please send me an Email and I will try to send a copy to Google Drive.

Hello, can you share the Google Drive download link,？ I can't download it by “tools/download_pretrained.sh“ either，Thanks!!!

kentang-mit, 2023-05-22:
Sorry for the delay. I just finished a conference deadline last week. Please have a look at [this link](https://drive.google.com/drive/folders/1Jru7VTfgvFF949DlP1oDfriP3wrmOd3c?usp=share_link) to download all the checkpoints.

closed 2023-07-15 by zhijian-liu

ldtho, 2023-09-29:
Hi @kentang-mit , I applied the fix in the GT box center ( origin=(0.5, 0.5, 0) to  origin=(0.5, 0.5, 0.5)), now can I use your checkpoint to evaluate, is it still compatible or do I need to re-train a new model? 
Thank you