mit-han-lab/bevfusion #400: An error occurred in the infer of Segmentation

Isaac-xie, 2023-04-28:
I I have reproduced the inference for the 3D Object Detection results, but when I try to reproduce the infer of Segmentation using the following code, I get an error. Incidentally I also get errors when using the training code for both detection and segmentation.
`CUDA_VISIBLE_DEVICES=4,5,6,7 torchpack dist-run -np 4 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`

Here is the error message

```
2023-04-27 09:06:59,200 - mmdet - INFO - load checkpoint from http path: https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth
load checkpoint from local path: pretrained/bevfusion-seg.pth
[                                                  ] 0/6019, elapsed: 0s, ETA:/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
[>>] 6020/6019, 17.6 task/s, elapsed: 342s, ETA:     0sKilled
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[38366,1],0]
  Exit code:    137
```

Here's my notes for setting up the environment

```
conda create -n liu_bevf python=3.8
conda activate liu_bevf
pip install torch==1.10.0+cu113 torchvision==0.11.0+cu113 -f https://download.pytorch.org/whl/torch_stable.html

pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html
pip install mmdet==2.20.0
pip install Pillow==8.4.0
pip install tqdm
pip install torchpack
pip install nuscenes-devkit
pip install setuptools==59.5.0
conda install openmpi -y

export OMPI_MCA_opal_cuda_support=true

conda install mpi4py -y
pip install ninja
pip install numba

python setup.py develop
```

kentang-mit, 2023-05-02:
Hi @Isaac-xie,

The environment setup looks pretty good to me, but I am not quite sure the reason that causes the error message. It could be the case that you do not have enough host memory. Did you have a chance to also try out inferencing with fewer number of GPUs and add a `-v` flag to your launching command?

Best,
Haotian

Isaac-xie, 2023-05-06:
@kentang-mit 
Thank you for your reply. 

I try to reproduce the infer of Segmentation using the new following code.
`CUDA_VISIBLE_DEVICES=0,1 torchpack dist-run -v -np 2 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`
But I still encountered errors

```
[>>] 6020/6019, 13.8 task/s, elapsed: 436s, ETA:     0s/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1838: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  for line in lines:
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1838: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  for line in lines:
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
/media/data2/xzz/software/anaconda3/envs/liu_bevf/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
  exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
[... cut ...]
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun noticed that process rank 1 with PID 0 on node arc-xzz8278-2djzf6-6fd7cc4f5b-dlmp6 exited on signal 9 (Killed).
```
I used two 4090 GPUs with 24G memory.

Best wishes,
isaac.

kentang-mit, 2023-05-22:
Sorry for the delayed response. It seems that your process got killed during the evaluation process. This is typically related to OOM. You may want to double check whether you ran out of the **host memory** (i.e. main memory, not the GPU memory) when running the evaluation code.

closed 2023-07-14 by zhijian-liu