mit-han-lab/bevfusion #402: depth estimation - different from the paper

lixinghe1999, 2023-04-30:
Dear author and people who are interested in this project.

From the paper: “we explicitly predict the discrete depth distribution of each pixel”, seems the depth is learned from rgb.
However, the implementation in "https://github.com/mit-han-lab/bevfusion/tree/main/mmdet3d/models/vtransforms/base.py" actually use points (from lidar) to get the depth.

kentang-mit, 2023-05-02:
Hi @lixinghe1999,

Yes, your understanding is correct, but I need to point out that the depth-assisted LSS (DLSS) transformation implemented in this codebase actually does not provide notable performance improvement over the LSS baseline, and it is obviously less efficient. This is the reason why we only mention we use LSS in the paper.

Best,
Haotian

TongNing-Albert, 2023-05-05:
> 

So why do you choose LSS to generate bev features in camera modality?

kentang-mit, 2023-05-22:
Hi Tong,

The reason why we choose LSS is that it is much more efficient than other alternatives with a good CUDA implementation and the performance is decent.

Best,
Haotian

closed 2023-07-15 by zhijian-liu