mit-han-lab/bevfusion #404: Some questions about `ImgAug3D` and the implementation based on new coordinate system of mmdet3d

JingweiZhang12, 2023-05-05:
1. I noticed that all transforms for images in `ImgAug3D` are not applied to ground-truth 3d boxes and these transforms are reverted when project image features to BEV space. I wonder how much the performance gain of  `ImgAug3D` is.
2. The rotation in `PIL` is counterclockwise by default. However, the rotation matrix [here](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/pipelines/transforms_3d.py#L83) is clockwise. Why?
3. I found that the performance of pretrained lidar detector drops when adding camera modality into training. Could you give me some advice? Here is the implementation of BEVFusion based in mmdet3d dev-1.x branch. https://github.com/JingweiZhang12/mmdetection3d/tree/bevfusion_ours. If possible, could you please add my WeChat: zjw209624

HuangVictorAuto, 2023-05-10:
> 1. The rotation in `PIL` is counterclockwise by default. However, the rotation matrix [here](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/pipelines/transforms_3d.py#L83) is clockwise. Why?
> 2. I found that the performance of pretrained lidar detector drops when adding camera modality into training. Could you give me some advice? Here is the implementation of BEVFusion based in mmdet3d dev-1.x branch. https://github.com/JingweiZhang12/mmdetection3d/blob/bevfusion_ours/projects/BEVFusion/configs/bevfusion_voxel0075_second_secfpn_8xb4-cyclic-20e_nus-3d.py. If possible, could you please add my WeChat: zjw209624

@JingweiZhang12 ， Hi, we have encountered the same problem, when add camera branch, the performance drops compared to only lidar branch.

as to rotation, I understand it is correct here.

PIL is default counterclockwise.

For a image coordinate, x to right, y to down. the default clockwise rotation matrix is
```
[[np.cos(theta), -np.sin(theta)],
[np.sin(theta), np.cos(theta)],]

```
and the corresponding counterclockwise rotation matrix is 
```
[[np.cos(theta), np.sin(theta)],
[-np.sin(theta), np.cos(theta)],]
```

JingweiZhang12, 2023-05-10:
@HuangVictorAuto Thanks. Indeed, the matrix is correct here. I found the lidar-camera model can converge in the training when removing `ImgAug3D`. I guess it may have something that needs to be modified correspondingly with the coordinates.

closed 2023-05-23 by JingweiZhang12