mit-han-lab/bevfusion #405: tensorboard logs

VeeranjaneyuluToka, 2023-05-05:
Hi All, i have been experimenting with custom dataset and i can not see the tensorboard train or validation logs. I could only see LR and Momentum which are provided by mmdet i think. Wondering if i missed anything to get tensorboard updated with train logs and if it is so where i should include that? Thanks!

kentang-mit, 2023-05-22:
Hi @VeeranjaneyuluToka,

I did not use tensorboard in this project. In fact we used wandb internally, which can be enabled by adding these lines to your `tools/train.py` after you set `cfg.run_dir`:

```python
if dist.is_master():
    import wandb

    wandb.init(
        project="PROJECT NAME HERE",
        entity="ENTITY NAME HERE",
        config=cfg._cfg_dict,
        sync_tensorboard=True,
    )
```

You can see if more things showed up on wandb. By the way, for me I usually only look at whether the losses go down in an expected speed and the validation accuracy numbers. For the models using the transfusion head, looking at the `matched_ious` can also be helpful. For other statistics I generally do not care too much about them in this project. So ideally even if you do not have the tensorboard you could still monitor the training process by looking at the `.log` file generated by mmdet3d.

Best,
Haotian

VeeranjaneyuluToka, 2023-05-24:
@kentang-mit , Thanks for your reply and will analyse the way you mentioned. However i am wondering how i am getting tensorboard logs when i train with NuScenes samples, here is the one that i get.
[image]

VeeranjaneyuluToka, 2023-05-26:
I could solve this issue myself. This is due to interval value in log_config in config file. Also i was getting the loss tensors in tensorboard, however i will also start using the wanddb tool, thanks for your kind reply.

closed 2023-05-26 by VeeranjaneyuluToka