mit-han-lab/bevfusion #408: mAP decrease to zero in the final epoch

BoomSky0416, 2023-05-10:
I used centerhead as bbox_head for the fusion model. And the LR is 1e-4. The loss curve and grad_norm curve are normal. Here are the mAP/loss/grad_norm curve. Has anyone else experienced the same situation and what could be the cause?
[image]
[image]
[image]

kentang-mit, 2023-05-22:
Hi @BoomSky0416,

In general the loss seems to be a bit higher than I expected. I usually expect something less than 2 even at the start. One possible reason is that you did not pretrain the LiDAR-only model before joint finetuning. In this case, the performance could be suboptimal and the training might not be stable. Would you mind double-checking whether it is the case?

Best,
Haotian

BoomSky0416, 2023-05-23:
Hello Haotian,

Thanks for your reply! 

Actually, I pre-train the LiDAR-only model before joint fine-tuning. But the difference is that I use CenterHead for LiDAR-only pre-training and joint fine-tuning. I don't know if the problem comes from here. What do I need to adjust to fit this CenterHead? And here are my LR curve and data augmentations used in joint fine-tuning. 

Thanks for your help!

[image]
`train_multiview_transforms = [
    dict(
        type='mmengine.RandomResize',
        scale=(1600, 900),
        ratio_range=(0.38, 0.55),
        keep_ratio=True,
        resize_type='mmdet.Resize',
        backend='pillow',
        interpolation='bicubic'),
    dict(
        type='RangeLimitedRandomCrop',
        crop_size=(704, 256),
        crop_type='absolute',
        relative_x_offset_range=(0.0, 1.0),
        relative_y_offset_range=(1.0, 1.0)),
    dict(type='mmdet.Pad', size=(704, 256)),
    dict(type='mmdet.RandomFlip', prob=0.5, direction='horizontal'),
    dict(
        type='mmdet.Rotate',
        prob=1.0,
        reversal_prob=0.5,
        level=None,
        min_mag=0.0,
        max_mag=5.4,
        img_border_value=0)
]

train_pipeline = [
    dict(
        type='LoadMultiViewImageFromFiles',
        num_views=6,
        set_default_scale=False),
    dict(
        type='mmdet3d.LoadPointsFromFile',
        coord_type='LIDAR',
        load_dim=5,
        use_dim=5),
    dict(
        type='mmdet3d.LoadPointsFromMultiSweeps',
        sweeps_num=9,
        use_dim=[0, 1, 2, 3, 4],
        pad_empty_sweeps=True,
        remove_close=True,
        backend_args=backend_args),
    dict(
        type='LoadAnnotations3D',
        with_bbox_3d=True,
        with_label_3d=True,
        with_attr_label=False),
    dict(
        type='MultiViewWrapper',
        process_keys=['img'],
        collected_keys=['img_shape', 'pad_shape', 'homography_matrix'],
        override_aug_config=False,
        transforms=train_multiview_transforms),
    dict(
        type='GlobalRotScaleTrans',
        rot_range=[-0.78539816, 0.78539816],
        scale_ratio_range=[0.9, 1.1],
        translation_std=0.5),
    dict(
        type='RandomFlip3D',
        sync_2d=False,
        flip_ratio_bev_horizontal=0.5,
        flip_ratio_bev_vertical=0.5,
        flip_box3d=True,
        flip_2d=False),
    dict(type='GetTransformationMatrices'),
    dict(
        type='ComputeDepthFromPoints',
        downsample=16,
        gt_type='cls',
        cls_mode='one_hot',
        d_min=0.5,
        d_max=59.5,
        dbins=59,
        dmode='LID'),
    dict(type='mmdet3d.PointsRangeFilter', point_cloud_range=point_cloud_range),
    dict(
        type='mmdet3d.ObjectRangeFilter',
        point_cloud_range=point_cloud_range),
    dict(
        type='mmdet3d.ObjectNameFilter',
        classes=[
            'car', 'truck', 'construction_vehicle', 'bus', 'trailer',
            'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'
        ]),
    dict(
        type='GridMask_bevfusion',
        use_h=True,
        use_w=True,
        max_epoch=6,
        rotate=1,
        offset=False,
        ratio=0.5,
        mode=1,
        prob=0.0,
        fixed_prob=True),
    dict(type='mmdet3d.PointShuffle'),
    dict(
        type='Pack3DDetInputs',
        keys=[
            'points', 'img', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_depth_dist',
            'gt_depth_valid_mask', 'cam2img', 'lidar2cam', 'img2img_aug',
            'lidar2lidar_aug', 'lidar_aug2img_aug', 'lidar2global'
        ],
        meta_keys=['img_shape', 'sample_idx', 'num_views'])
]`

HydrogenWasser, 2023-11-24:
> Hello Haotian,
>
> Thanks for your reply!
>
> Actually, I pre-train the LiDAR-only model before joint fine-tuning. But the difference is that I use CenterHead for LiDAR-only pre-training and joint [...]

hi，请问你有尝试过训练超过6个epoch么，我在自己的数据集上，前6个epochloss正常下降，但从第七个epoch开始loss开始上升，mAP下降，最终直到0.

closed 2024-05-04 by zhijian-liu