mit-han-lab/bevfusion #409: Evaluation processing time

SMSajadi99, 2023-05-14:
Hello
I'm testing the `evaluation `stage, but after downloading the checkpoint for Torch, it waits here and does nothing?
Is there a problem or should I really wait?

[image]

SMSajadi99, 2023-05-14:
[image]

[image]

SMSajadi99, 2023-05-18:
How much time is needed to do the evaluation?

VanderBieu, 2023-05-19:
I had similar issue during evaluation. You can try reducing number of nodes by change `torchpack -np 8` to `torchpack -np 1` It works for me.
So it appears that the number of nodes must match the number of GPUs.

SMSajadi99, 2023-05-19:
thanks so much

closed 2023-05-19 by SMSajadi99