mit-han-lab/bevfusion #412: flops counter problem

ycwang1997, 2023-05-15:
Hi, I want to measure the #MACs. I have referred to issues #8 and #298 .
I am trying to inject the `flops_counter` function to `benchmark.py`.
The code is shown below:
```
# benchmark with several samples and take the average
    for i, data in enumerate(data_loader):

        torch.cuda.synchronize()
        start_time = time.perf_counter()

        with torch.no_grad():
            macs, params = flops_counter(model, **data)
            print('macs:', macs)
            print('params:', params)
            model(return_loss=False, rescale=True, **data)

        torch.cuda.synchronize()
        elapsed = time.perf_counter() - start_time
```
However, a typeerror occurred.
[image]

Can you provide me with some suggestions to assist me in measuring the #MACs?
Thank you.

kentang-mit, 2023-05-22:
Hi @ycwang1997,

I checked the code I previously used to call this `flops_counter`. Maybe it is helpful to use:

```python
macs, params = flops_counter(model, data)
```

instead of 

```python
macs, params = flops_counter(model, **data)
```

Please do not hesitate to contact me again if you still meet with errors and feel free to include a longer error log.

Best,
Haotian

ycwang1997, 2023-05-25:
Hi @kentang-mit, 

Thank you for the suggestion.
I have tried that instruction before.
However, the same type error still occurred.
This problem occurs whether I use the model I trained or your pre-trained model.

Could you calculate the #macs on the pre-trained model with this instruction?

wulinaaa, 2023-11-27:
Hello, have you solved this problem？

qq1018408006, 2024-04-03:
@kentang-mit Thans for your reply! I additionally wonder if the bev pooling operation is considered when countering model MACs.

closed 2024-05-04 by zhijian-liu