mit-han-lab/bevfusion #413: Dataloader weird effect on memory

polo5, 2023-05-18:
Heya, thanks for this great repo.

I've noticed the CPU memory varies a lot during training. It increases during each training epoch slowly, and it increases massively from the first time we do validation. In the below runs you can see how the System memory increases as soon as I do validation for the first time on epoch 5: 

[image]

To make things worse, both of the runs above use the same model and dataloader config (namely 4gpus, 12 workers per gpu). I observed the same behavior for 6-cam and 1-cam models.

Any idea for this weird behaviour from your data pipeline?

kentang-mit, 2023-05-22:
Hi @polo5,

Thanks for the interest. I had a look at some of my previous runs. The system memory utilization typically looks like this. I did not observe too many large spikes in the log. One difference is that I am using 4 workers per GPU for 8-GPU workloads. Notice that the CPU I am using have 96 threads when hyperthreading is turned on. 

[image]

Best,
Haotian

polo5, 2023-05-23:
Thanks, I found that it only happens when you use all the cores available (e.g. 12 workers per GPU for a 4 GPU machine with 48 cores). The solution is to leave at least 1-2 cores free. Not sure if this is a mmcv issue or a pytorch DDP issue

closed 2023-05-23 by polo5