mit-han-lab/bevfusion #414: Lidar branch Network pretrained ckpt

imyyf, 2023-05-19:
In cam branch, the model has pretrained ResNet ckpt. Why Lidar branch doesn't have/use pretrained ckpt.

kentang-mit, 2023-05-22:
Hi @imyyf,

Actually we did not use ResNet pretraining for the camera branch. Instead we are using SwinT. For the LiDAR branch, if you are using the fusion model, we actually also utilized the pretrained LiDAR-only checkpoint (see README.md for instructions).

Best,
Haotian

imyyf, 2023-05-26:
Thanks for your reply. 
Sorry for the wrong question about cam branch.
Actually, I'm curious about that.
We find that on cam branch it has pretrained ckpts for training.
But on lidar branch, it doesn't.
Is there any obstacle for lidar branch pretrained ckpts for training. Or it is trivial, need not do that.

closed 2024-05-04 by zhijian-liu