mit-han-lab/bevfusion #416: How can i visualize result of test?

SeungHwi0613, 2023-05-22:
I wanted to test(no train no evaluation) my custom dataset so i made it to same format like NuScenes. 
->So obviously, i made everything only except few files related with Annotation (instance.json / sample_data_annotation.json)
After that process, I success to make test.pkl and test_mono3d.json file.

And then i start to test my dataset with

**torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml pretrained/camera-only-det.pth --format-only --eval-options 'jsonfile_prefix=./result/result.json'**

and then i get

Traceback (most recent call last):
  File "tools/test.py", line 232, in <module>
    main()
  File "tools/test.py", line 205, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/mmdet/apis/test.py", line 96, in multi_gpu_test
    for i, data in enumerate(data_loader):
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 359, in __iter__
    return self._get_iterator()
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 305, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 944, in __init__
    self._reset(loader, first_iter=True)
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 975, in _reset
    self._try_put_index()
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1209, in _try_put_index
    index = self._next_index()
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 512, in _next_index
    return next(self._sampler_iter)  # may raise StopIteration
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/torch/utils/data/sampler.py", line 229, in __iter__
    for idx in self.sampler:
  File "/home/hwi/anaconda3/lib/python3.8/site-packages/mmdet/datasets/samplers/distributed_sampler.py", line 35, in __iter__
    math.ceil(self.total_size / len(indices)))[:self.total_size]
ZeroDivisionError: division by zero
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[64015,1],0]
  Exit code:    1
--------------------------------------------------------------------------

this error.

i don't know why 'len(indices)' is zero

kentang-mit, 2023-05-22:
Hi @SeungHwi0613,

I had a look at `cfg.data.test`, it seems that it is a dictionary (specified [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/configs/nuscenes/default.yaml#L286)). The reason why its length is 9 is that we have 9 keys. So it is not representing the actual size of the constructed dataset. 

Regarding the division by zero error, I am not sure how you defined the dataset, but it will be helpful if you can double check whether the `__len__` function is defined for your dataset. For example, if your dataset is inherited from [Custom3DDataset](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/custom_3d.py#L16), then the `__len__` function is defined [here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/custom_3d.py#L267).

Best,
Haotian

SeungHwi0613, 2023-05-23:
__len__ also return 'len(self.data_infos)' this value
so
when i tried 

'cfg.data.test.__len__()'->9 return
'dataset.__len__()'0 return

still same
---
and when i trace back to the error, mmdet/apis/test.py line70 'enumerate(data_loader)' was occuring the error i guess. 

It would be so nice if you give me some advice about this error

i really appreciate about your answer. 

sincerely

closed 2023-06-19 by SeungHwi0613

AlexandreDiPiazza, 2024-06-06:
Hey SeungHwi0613, 

I am also interested in using the code (both training and testing) on my custom dataset - could you please provide some details on the steps you made in your case? 

Thank you !