mit-han-lab/bevfusion #417: How to visualize the result of test?

SeungHwi0613, 2023-05-22:
Hi

I just use NuScenes v1.0-test for test 
and i commented out everything about annotation and GT 
so i success to made pkl and mono3d.json

After that, when i use

**torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml pretrained/camera-only-det.pth --format-only --eval-options 'jsonfile_prefix=./result/result.json'**

this command to save result as a '.json' format file.

But i want to visualize result of test like evaluation(i did it before as your instruction)

How can i visualize the result of test?

SeungHwi0613, 2023-05-23:
But when i tried this command before ti said i need 'val.pkl'. that's why i upload this question. how can i fix Visualiza.py code fir on test?

sandeepnmenon, 2023-06-02:
While running the above command, I am getting the error

```
0% 0/81 [00:02<?, ?it/s]
Traceback (most recent call last):
  File "tools/visualize.py", line 166, in <module>
    main()
  File "tools/visualize.py", line 88, in main
    outputs = model(**data)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 230, in forward_single
    feature = self.extract_camera_features(
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 114, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/models/vtransforms/lss.py", line 76, in forward
    x = super().forward(*args, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func
    return old_func(*args, **kwargs)
TypeError: forward() takes 11 positional arguments but 12 were given
--------------------------------------------------------------------------
```

I printed the keys of the `data` variable that is being passed and it has
` dict_keys(['img', 'points', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_masks_bev', 'camera_intrinsics', 'camera2ego', 'lidar2ego', 'lidar2camera', 'camera2lidar', 'lidar2image', 'img_aug_matrix', 'lidar_aug_matrix', 'metas'])`

sandeepnmenon, 2023-06-02:
I had to add a new argument for `metas` here
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/vtransforms/base.py#L177-L178

because that is being passed here
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/fusion_models/bevfusion.py#L125

Not sure if it is a fix, but I am running
```bash
torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml --mode pred --checkpoint pretrained/bevfusion-seg.pth
```
and it does not run without the fix I mentioned

SeungHwi0613, 2023-06-05:
i also got same error. so i just changed **kwargs -> kwargs
it is working but i also don't know it is okay or not

HenryZhangJianhe, 2023-07-14:
you can add metas option in mmdet3d/models/vtransforms/base.py
```python
def forward(
        self,
        img,
        points,
        sensor2ego,  #cam2ego
        lidar2ego,
        lidar2camera,
        lidar2image,
        cam_intrinsic,
        camera2lidar,
        img_aug_matrix,
        lidar_aug_matrix,
        metas,
        **kwargs,
    ):
```

closed 2024-05-04 by zhijian-liu