mit-han-lab/bevfusion #418: Changing the point cloud range

VeeranjaneyuluToka, 2023-05-22:
Am trying to change the point cloud range from **[-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]** to **[-115.0, -74.0, 0.0, 230.0, 320.0, 30.0]**, it fails with the below error

return forward_call(*input, **kwargs)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 119, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/ops/spconv/conv.py", line 197, in forward
    out_features = Fsp.indice_subm_conv(
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/cuda/amp/autocast_mode.py", line 92, in decorate_fwd
    return fwd(*_cast(args, cast_inputs), **_cast(kwargs, cast_inputs))
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/cuda/amp/autocast_mode.py", line 35, in _cast
    return type(value)(iterable)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/cuda/amp/autocast_mode.py", line 33, in <lambda>
    iterable = map(lambda v: _cast(v, dtype), value)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/cuda/amp/autocast_mode.py", line 25, in _cast
    return value.to(dtype) if is_eligible else value
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
[hykeserver:688058] *** Process received signal ***
[hykeserver:688058] Signal: Aborted (6)
[hykeserver:688058] Signal code:  (-6)
[hykeserver:688058] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x14420)[0x7f0655bf1420]
[hykeserver:688058] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7f0655a2e00b]
[hykeserver:688058] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7f0655a0d859]
[hykeserver:688058] [ 3] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/../../../../libstdc++.so.6(+0xb135a)[0x7f06510cf35a]
[hykeserver:688058] [ 4] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/../../../../libstdc++.so.6(+0xb03b9)[0x7f06510ce3b9]
[hykeserver:688058] [ 5] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/../../../../libstdc++.so.6(__gxx_personality_v0+0x87)[0x7f06510ceae7]
[hykeserver:688058] [ 6] /home/hykeserver/anaconda3/envs/bevf_ptt19/bin/../lib/libgcc_s.so.1(+0x111e4)[0x7f065273b1e4]
[hykeserver:688058] [ 7] /home/hykeserver/anaconda3/envs/bevf_ptt19/bin/../lib/libgcc_s.so.1(_Unwind_Resume+0x12e)[0x7f065273bc1e]
[hykeserver:688058] [ 8] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/libc10_cuda.so(+0x1d2fb)[0x7f064badf2fb]
[hykeserver:688058] [ 9] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/libc10.so(_ZN3c1010TensorImpl17release_resourcesEv+0xa4)[0x7f064b865314]
[hykeserver:688058] [10] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/libtorch_python.so(+0x295359)[0x7f05e913b359]
[hykeserver:688058] [11] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/libtorch_python.so(+0xadb231)[0x7f05e9981231]
[hykeserver:688058] [12] /home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/lib/libtorch_python.so(_Z28THPVariable_subclass_deallocP7_object+0x292)[0x7f05e9981532]
[hykeserver:688058] [13] python(+0x11cb86)[0x563e9b7b7b86]
[hykeserver:688058] [14] python(+0x11cf76)[0x563e9b7b7f76]
[hykeserver:688058] [15] python(+0x11cf76)[0x563e9b7b7f76]
[hykeserver:688058] [16] python(+0x1109a0)[0x563e9b7ab9a0]
[hykeserver:688058] [17] python(+0x11d346)[0x563e9b7b8346]
[hykeserver:688058] [18] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [19] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [20] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [21] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [22] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [23] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [24] python(+0x11d2fc)[0x563e9b7b82fc]
[hykeserver:688058] [25] python(PyDict_SetItem+0x2ac)[0x563e9b80025c]
[hykeserver:688058] [26] python(PyDict_SetItemString+0x4f)[0x563e9b80094f]
[hykeserver:688058] [27] python(PyImport_Cleanup+0x9b)[0x563e9b8e546b]
[hykeserver:688058] [28] python(Py_FinalizeEx+0x83)[0x563e9b8e5823]
[hykeserver:688058] [29] python(Py_RunMain+0x110)[0x563e9b8e8780]
[hykeserver:688058] *** End of error message ***
Aborted (core dumped)

I noticed that my GPU memory is not used completely, anybody faced this issue?

kentang-mit, 2023-05-22:
Hi @VeeranjaneyuluToka,

One potential reason why you met with this error is that there exists a parity between the `point_cloud_range` and `grid_size` ([here](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/configs/nuscenes/det/transfusion/default.yaml#L20)) in the head. For example, if we have xy range = [-51.2m, 51.2m] and voxel_size=0.1m, the `grid_size` should be [1024, 1024] because (51.2-(-51.2)) / 0.1 = 1024. Therefore, you may want to adjust these `grid_size` parameters accordingly. These parameters are used in SparseTensor construction and a too small value may lead to negative coordinates, which breaks the neighborhood query kernel in the SpConv backend.

Best,
Haotian

VeeranjaneyuluToka, 2023-05-23:
@kentang-mit , Thanks for your reply!

I changed as below

voxel_size = [0.075, 0.075, 0.2] - no change
point_cloud_range = [-400.0, -400.0, -78.04, 400.0, 400.0, 78.04]
grid_size = [ 10667, 10667, 1]

with the above settings, am getting the error as below
 File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 251, in forward_single
    feature = self.extract_lidar_features(points)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 139, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 119, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/home/hykeserver/Downloads/git_repos/object_detection_3d/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: mmdet3d/ops/spconv/src/indice_cuda.cu 124
cuda execution failed with error 2
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.

Any inputs on this?

VeeranjaneyuluToka, 2023-05-23:
@kentang-mit , i experimented a bit more on the valid range, changed accordingly and it solved this issue.  Thanks!

closed 2023-05-23 by VeeranjaneyuluToka