mit-han-lab/bevfusion #422: result of custom dataset test has wrong direction

SeungHwi0613, 2023-05-25:
My result of custom dataset test has wrong direction.

I guess it is a problem of sensor_calibration.
But before i start to fix this problem i just want to make sure.
(first, i made my dataset to same format like nuscenes.)

1) this result is caused by calibration? or also ego_pose?
2) I don't need map segmentation that's why i just put [ego_pose] translation to zero. Is it also can be the reason?

[image]
[image]
[image]

sandeepnmenon, 2023-06-02:
How did you write the create_data pipeline for your custom dataset? I am looking to evaluate this model on my custom data.

SeungHwi0613, 2023-06-05:
I just made my dataset same format like nuScenes

Chen-hai-feng, 2023-08-01:
> 

Hello, are you directly generating pkl in the same format as nuscenes?

SeungHwi0613, 2023-08-01:
> > 
> 
> Hello, are you directly generating pkl in the same format as nuscenes?

Yes, i made it in same format as nuscenes

Chen-hai-feng, 2023-08-01:
> 

Well, this is the case. I recently encountered some problems when transcribing my own data into nuscenes, 6 cameras and a laser. Can you provide the following reference?

AlexIlis, 2023-08-25:
@SeungHwi0613 Since this repository uses Nuscenes based metadata, how did you generate the custom data pkl files in the same format as that of NuScenes ?

YangChen1234567, 2023-09-07:
@SeungHwi0613 Hi, I am trying BEVFusion in custom dataset too(on single image). But I met a performance drop after fusion, have you met this problem??? 
Looking forward to your reply... Thanks..

evil-master, 2023-10-26:
> 嗨，我也在自定义数据集中尝试 BEVFusion（在单个图像上）。但是我遇到了融合后性能下降，你遇到过这个问题吗???期待您的回复...谢谢。。
请问你自定义的数据集是转换成nusence数据格式实现的吗？还是kitti数据格式实现的？

YangChen1234567, 2023-10-26:
> > 嗨，我也在自定义数据集中尝试 BEVFusion（在单个图像上）。但是我遇到了融合后性能下降，你遇到过这个问题吗???期待您的回复...谢谢。。
> > 请问你自定义的数据集是转换成nusence数据格式实现的吗？还是kitti数据格式实现的？

nuscenes格式。

evil-master, 2023-10-26:
请问是如何将自己的数据制作成nuscenes数据格式的呢？（目前我只有一个lidar和一个相机和基于kitti数据集格式的标注）如果将自己标注的数据转成nusences数据格式后需要自己重新写生成pkl文件吗？因为我看它的pkl生成时调用了NuScenes库去处理的，所以我将数据整合成nusences格式一样后能被这个库正常处理吗？

大力金刚
***@***.***

YangChen1234567, 2023-10-26:
自己写脚本。

evil-master, 2023-10-26:
好的，是数据处理、以及后续的网络，那部分自己重写读取并生成pkl吗？还需要将自己的数据集格式转成和nusences数据集格式一样吗？

大力金刚
***@***.***