mit-han-lab/bevfusion #424: TTA and model ensembling

NoOneUST, 2023-05-27:
Thank you for your great work. Did you provide any tutorial or code on how to do TTA and model ensembling? It seems that these techniques benefit the effectiveness a lot.

jamesgunnfiveai, 2023-06-09:
I would be very interested in this information as well if it's able to be provided. Concrete questions I have about it are:
* Ablation of the TTA strategies (double flip, rotation, ensembling)
* Do you concatenate the different strategy classes or take the cartesian product?
* For your best nuScenes results did you use circlenms? Did you run the NMS on each class separately? If so, did you use class specific thresholds and what were they?

zhijian-liu, 2024-05-04:
We follow the standard test-time augmentation, as many other methods on the leaderboard have done. We haven't released the implementation of this part yet. We have recently shifted focus to different research directions, so unfortunately, we don't have enough bandwidth to clean up the code for this.

closed 2024-05-04 by zhijian-liu