mit-han-lab/bevfusion #426: why ObjectPaste had about 1map improvement for camera_only branch?

U201310646, 2023-05-30:
Hey thanks for great repo first,

I noticed that the config in the camera branch training process still retains lidar-related operations such as ObjectPaste. Although no point cloud information is used in the camera training, the addition of gt_boxes and gt_labels by ObjectPaste still has some impact on the training. In my experiments, I found that the map could reach 36.2 when ObjectPaste was present, and dropped to 35.2 without ObjectPaste. However, theoretically, the ObjectPaste operation added extra meaningless target boxes to the images, which could not be matched with the image features. But why such a result occurs and could u please explain this? Btw why u keep ObjectPaste in camera only branch?

Best
Rui

U201310646, 2023-06-12:
one more question: why change center_int to center_int[[1,0]] in camera only branch https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/heads/bbox/centerpoint.py#L548

U201310646, 2023-06-13:
> one more question: why change center_int to center_int[[1,0]] in camera only branch
> 
> https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/heads/bbox/centerpoint.py#L548

this problem i figure it out. could u pls the first question explain? ty

zhijian-liu, 2024-05-04:
Unfortunately, I don't have an explanation for this phenomenon. It may be due to fluctuations during training.

closed 2024-05-04 by zhijian-liu