mit-han-lab/bevfusion #428: Segmentation error when running without torchpack and distributed mode

sandeepnmenon, 2023-06-06:
I noticed that the following function call 
https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/ops/voxel/voxelize.py#L55-L66
throws a segmentation error if I am not running with torchpack.
```bash
(base) root@abf60d2d48ce:/home/bevfusion# python tools/visualize.py /home/bevfusion/configs/nuscenes/seg/lidar-centerpoint-bev128.yaml --mode pred --checkpoint pretrained/lidar-only-seg.pth --split val                    
load checkpoint from local path: pretrained/lidar-only-seg.pth
0it [00:00, ?it/s]Segmentation fault (core dumped)
```

I narrowed down the bug to the hard_voxelise call above but could not get a trace

PS: I made some modifications in the visualize.py code to put the distributed code (like the init and MMDistributedDataParallel) inside an argument check.

GoodarzMehr, 2023-12-28:
Hi,

Did you figure out how to resolve this?

GoodarzMehr, 2023-12-28:
I noticed that when using torchpack the tensors are placed on the GPU memory, but without it (also commenting out `dist.init()` and `MMDistributedDataParallel` stuff) they are placed on the CPU memory, and there seems to be a bug in the `hard_voxelize()` CPU code. What ended up working for me was manually placing the inputs on the GPU memory, i.e. doing this in the `bevfusion.py` file:

```python3
@auto_fp16(apply_to=("img", "points"))
    def forward(
        self,
        img,
        points,
        camera2ego,
        lidar2ego,
        lidar2camera,
        lidar2image,
        camera_intrinsics,
        camera2lidar,
        img_aug_matrix,
        lidar_aug_matrix,
        metas,
        gt_masks_bev=None,
        gt_bboxes_3d=None,
        gt_labels_3d=None,
        **kwargs,
    ):
        if isinstance(img, list):
            raise NotImplementedError
        else:
            outputs = self.forward_single(
                img.cuda(),
                points.cuda(),
                camera2ego.cuda(),
                lidar2ego.cuda(),
                lidar2camera.cuda(),
                lidar2image.cuda(),
                camera_intrinsics.cuda(),
                camera2lidar.cuda(),
                img_aug_matrix.cuda(),
                lidar_aug_matrix.cuda(),
                metas,
                gt_masks_bev,
                gt_bboxes_3d,
                gt_labels_3d,
                **kwargs,
            )
            return outputs
```
Hope this helps.

closed 2024-05-04 by zhijian-liu