mit-han-lab/bevfusion #449: Cannot Import from Partially Initialized Module (most likely due to a circular import)

NoOneUST, 2023-07-09:
```
(base) torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
Traceback (most recent call last):
  File "tools/test.py", line 15, in <module>
    from mmdet3d.datasets import build_dataloader, build_dataset
  File "/export/data/mine/bevfusion/mmdet3d/datasets/__init__.py", line 4, in <module>
    from .custom_3d import *
  File "/export/data/mine/bevfusion/mmdet3d/datasets/custom_3d.py", line 10, in <module>
    from ..core.bbox import get_box_type
  File "/export/data/mine/bevfusion/mmdet3d/core/__init__.py", line 2, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/export/data/mine/bevfusion/mmdet3d/core/bbox/__init__.py", line 3, in <module>
    from .iou_calculators import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/export/data/mine/bevfusion/mmdet3d/core/bbox/iou_calculators/__init__.py", line 1, in <module>
    from .iou3d_calculator import (
  File "/export/data/mine/bevfusion/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 5, in <module>
    from ..structures import get_box_type
  File "/export/data/mine/bevfusion/mmdet3d/core/bbox/structures/__init__.py", line 1, in <module>
    from .base_box3d import BaseInstance3DBoxes
  File "/export/data/mine/bevfusion/mmdet3d/core/bbox/structures/base_box3d.py", line 5, in <module>
    from mmdet3d.ops.iou3d import iou3d_cuda
  File "/export/data/mine/bevfusion/mmdet3d/ops/__init__.py", line 12, in <module>
    from .feature_decorator import feature_decorator
  File "/export/data/mine/bevfusion/mmdet3d/ops/feature_decorator/__init__.py", line 1, in <module>
    from .feature_decorator import feature_decorator
  File "/export/data/mine/bevfusion/mmdet3d/ops/feature_decorator/feature_decorator.py", line 4, in <module>
    from mmdet3d.ops.feature_decorator import feature_decorator_ext
ImportError: cannot import name 'feature_decorator_ext' from partially initialized module 'mmdet3d.ops.feature_decorator' (most likely due to a circular import) (/export/data/mine/bevfusion/mmdet3d/ops/feature_decorator/__init__.py)
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[64666,1],7]
  Exit code:    1
--------------------------------------------------------------------------
```

Isaac-xie, 2023-07-11:
Hello, I have the same problem.
May I ask how you solved it.

1145520074, 2023-07-11:
Hello, I have the same problem.
May I ask how you solved it.

Gss-DUT, 2023-07-11:
I also have the same problem,could you please tell me how to solve it

For24, 2023-07-13:
Hi, I found a quick but dirty workaround. 

I found that 'feature_decorator' is a new module in latest update, so I simply comment the import of this module in 'mmdet3d/ops/\_\_init\_\_.py' like this:

```py
...
from .ball_query import ball_query
# from .feature_decorator import feature_decorator
from .furthest_point_sample import (
    Points_Sampler,
    furthest_point_sample,
    furthest_point_sample_with_dist,
)
...
```
And then I could run the code. 

So I guess if your code doesn't use this model: [mmdet3d/models/backbones/radar_encoder.py](https://github.com/mit-han-lab/bevfusion/blob/d0152cf97c0ee3c7999c3f2227ab33e6620cee8f/mmdet3d/models/backbones/radar_encoder.py#L15), this workaround would help a little.

Gss-DUT, 2023-07-14:
> Hi, I found a quick but dirty workaround.
> 
> I found that 'feature_decorator' is a new module in latest update, so I simply comment the import of this module in 'mmdet3d/ops/__init__.py' like this:
> 
> ```python
> ...
> from .ball_query import ball_query
> # from .feature_decorator import feature_decorator
> from .furthest_point_sample import (
>     Points_Sampler,
>     furthest_point_sample,
>     furthest_point_sample_with_dist,
> )
> ...
> ```
> 
> And then I could run the code.
> 
> So I guess if your code doesn't use this model: [mmdet3d/models/backbones/radar_encoder.py](https://github.com/mit-han-lab/bevfusion/blob/d0152cf97c0ee3c7999c3f2227ab33e6620cee8f/mmdet3d/models/backbones/radar_encoder.py#L15), this workaround would help a little.

Hi，I tried this method but still got an error. The command I'm running is ‘’torchpack dist-run -np 1 python tools/train.py  configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth‘’，but  it also  cannot import name 'feature_decorator_ext'.Could you please tell me your run command？or Do you know what else it might be，Thanks

For24, 2023-07-14:
> Hi，I tried this method but still got an error. The command I'm running is ‘’torchpack dist-run -np 1 python tools/train.py [...]

Hi, I meet this error when I run the command to prepare the data and after I comment the previous line I could run the code.

For training, what about also comment this line in 'mmdet3d/models/backbones/\_\_init\_\_.py':

```py
...
from .vovnet import *
from .dla import *
# from .radar_encoder import *
``
I guess there are bugs in this 'rader_encoder', so if you don't use it, just comment the import of it.

Gss-DUT, 2023-07-14:
Thanks for your reply

> > > Hi, I found a quick but dirty workaround.
> > > I found that 'feature_decorator' is a new module in latest update, so I simply comment the import of this module in 'mmdet3d/ops/**init**.py' like this:
> > > ```python
> > > ...
> > > from .ball_query import ball_query
> > > # from .feature_decorator import feature_decorator
> > > from .furthest_point_sample import (
> > >     Points_Sampler,
> > >     furthest_point_sample,
> > >     furthest_point_sample_with_dist,
> > > )
> > > ...
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > And then I could run the code.
> > > So I guess if your code doesn't use this model: [mmdet3d/models/backbones/radar_encoder.py](https://github.com/mit-han-lab/bevfusion/blob/d0152cf97c0ee3c7999c3f2227ab33e6620cee8f/mmdet3d/models/backbones/radar_encoder.py#L15), this workaround would help a little.
> > 
> > 
> > Hi，I tried this method but still got an error. The command I'm running is ‘’torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth‘’，but it also cannot import name 'feature_decorator_ext'.Could you please tell me your run command？or Do you know what else it might be，Thanks
> 
> Hi, I meet this error when I run the command to prepare the data and after I comment the previous line I could run the code.
> 
> For training, what about also comment this line in 'mmdet3d/models/backbones/__init__.py':
> 
> ```python
> ...
> from .vovnet import *
> from .dla import *
> # from .radar_encoder import *
> ```
> 
> I guess there are bugs in this 'rader_encoder', so if you don't use it, just comment the import of it.

Thanks for your reply

zhijian-liu, 2023-07-21:
@K-Shao, could you please take a look at this issue? Thanks!

K-Shao, 2023-07-22:
Sorry for the delay! This behavior is due to a bug in the feature decorator, which is a dummy OP for the ONNX exporting, which is not actually supported in the codebase currently. I have a PR (#466) that will fix the issue, but it should also simply work to (as @For24 mentions) comment out the feature decorator import in mmdet3d/ops/__init__.py (however, you should also comment lines 15 and 177-178 in radar_encoder.py).

closed 2023-07-23 by zhijian-liu

poddar414, 2023-07-29:
I am still seeing this issue.

jongchyisu, 2023-07-31:
Also facing the same issue!

sangjinpark97, 2023-08-24:
@poddar414 @jongchyisu Have you solved it?

Wangquans, 2024-02-02:
I have sloved this problem,you can  comment the import of this module in 'mmdet3d/ops/init.py' like this:
`from .ball_query import ball_query
#from .feature_decorator import feature_decorator
from .furthest_point_sample import (
    Points_Sampler,
    furthest_point_sample,
    furthest_point_sample_with_dist,
)`
then it still occurs another simliar problem,you should comment another  import of this module in 'mmdet3d/ops/feature_decorator/init.py' like this:
`#from .feature_decorator import feature_decorator`

then the problem is solved.

Wangquans, 2024-02-02:
I have sloved this problem,you can  comment the import of this module in 'mmdet3d/ops/init.py' like this:
`
#from .feature_decorator import feature_decorator`
then it still occurs another simliar problem,you should comment another  import of this module in 'mmdet3d/ops/feature_decorator/init.py' like this:
`#from .feature_decorator import feature_decorator`

then the problem is solved.