mit-han-lab/bevfusion #456: Question about create_data

pkqbajng, 2023-07-16:
Thanks for your effort for such an excellent job, however, when I use the function of creating data provided by this project, the following errors occurred:

(bevfusion) python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes --workers 32
======
Loading NuScenes tables for version v1.0-trainval...
Loading nuScenes-lidarseg...
Loading nuScenes-panoptic...
32 category,
8 attribute,
4 visibility,
64386 instance,
12 sensor,
10200 calibrated_sensor,
2631083 ego_pose,
68 log,
850 scene,
34149 sample,
2631083 sample_data,
1166187 sample_annotation,
4 map,
34149 lidarseg,
34149 panoptic,
Done loading in 35.345 seconds.
======
Reverse indexing ...
Done reverse indexing in 9.4 seconds.
======
total scene num: 850
exist scene num: 850
train scene: 700, val scene: 150
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] 34149/34149, 2.3 task/s, elapsed: 14803s, ETA:     0s
nuscenes
train sample: 28130, val sample: 6019
Create GT Database of NuScenesDataset
Traceback (most recent call last):
  File "/mine/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/mine/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/mine/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/mine/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 202, in load_annotations
    data = mmcv.load(ann_file)
  File "/mine/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/fileio/io.py", line 60, in load
    with BytesIO(file_client.get(file)) as f:
  File "/mine/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 992, in get
    return self.client.get(filepath)
  File "/mine/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 517, in get
    with open(filepath, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: './data/nuscenes/nuscenes_infos_train.pkl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/create_data.py", line 101, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 45, in nuscenes_data_prep
    create_groundtruth_database(
  File "/mine/bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database
    dataset = build_dataset(dataset_cfg)
  File "/mine/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/mine/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: './data/nuscenes/nuscenes_infos_train.pkl'

pkqbajng, 2023-07-16:
It seems that you generated this file using the other codes? I tried to generate it using mmdet3d of version 1.2.0, but the keys are different from those utilized in your dataloader.

hbc2020, 2023-07-16:
Hi,
I met the same problem and after some debugging, I find that line 95~100 of nuscenes_converter.py should be modified as follows in order to generate *.pkl files successfully,
info_path = osp.join(root_path,
                             '{}_infos_train.pkl'.format(info_prefix))
info_val_path = osp.join(root_path,
                                 '{}_infos_val.pkl'.format(info_prefix))
Now I can generate *.pkl files by running python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes.

pkqbajng, 2023-07-17:
> Hi, I met the same problem and after some debugging, I find that line 95~100 of nuscenes_converter.py should be modified as follows in order to generate *.pkl files successfully, info_path = [...]

Thanks, I got it.

closed 2023-07-17 by pkqbajng