mit-han-lab/bevfusion #458: RuntimeError

huzhihen, 2023-07-18:
RuntimeError: Given groups=1, weight of size [8, 1, 1, 1], expected input[24, 6, 256, 704] to have 1 channels, but got 6 channels instead.

huzhihen, 2023-07-18:
In the bevfusion/mmdet3d/vtransforms/depth_loss.py, I change the code on line 39 nn.Conv2d(1, 8, 1) -> nn.Conv2d(6, 8, 1)，and code can run.

huzhihen, 2023-07-21:
I find the performance on the complete dataset of nuscenes is also not expected.

JSC1375, 2023-07-21:
I try to rollback git version.  It may help.
After setting to previous version and reruning pip install -v -e . and create_data.py, it has no runtime error, and the performance is better on v1.0-mini, but is still not expected. 
[image]

How is the performance on complete dataset? Hope for your reply.

huzhihen, 2023-07-21:
The results I obtained on the complete dataset are as follows. The checkpoint is pretrained/bevfusion-det.pth.
```
Results writes to /tmp/tmpssweul6f/results/results_nusc.json
mAP: 0.3509                              
mATE: 0.3505
mASE: 0.3598
mAOE: 0.4404
mAVE: 0.4063
mAAE: 0.1946
NDS: 0.5003
Eval time: 102.1s

Per-class results:
Object Class	AP	ATE	ASE	AOE	AVE	AAE
car	0.226	0.388	0.300	0.321	0.629	0.197
truck	0.332	0.425	0.307	0.234	0.432	0.211
bus	0.492	0.338	0.215	0.119	0.549	0.250
trailer	0.157	0.605	0.353	0.752	0.294	0.135
construction_vehicle	0.144	0.666	0.420	0.973	0.118	0.340
pedestrian	0.673	0.179	0.379	0.462	0.289	0.129
motorcycle	0.016	0.278	0.377	0.418	0.684	0.279
bicycle	0.350	0.169	0.356	0.483	0.255	0.015
traffic_cone	0.638	0.165	0.478	nan	nan	nan
barrier	0.481	0.293	0.413	0.201	nan	nan
```

huzhihen, 2023-07-21:
I would like to ask which version of code you rollback？

JSC1375, 2023-07-21:
the previous one. 
commit db75150717a9462cb60241e36ba28d65f6908607 (HEAD -> main)
Author: Zhijian Liu <zhijianliu.cs@gmail.com>
Date:   Thu Feb 9 16:02:55 2023 -0500

    Update README.md

> I would like to ask which version of code you rollback？

huzhihen, 2023-07-22:
I use the newest code and change
mmdet3d/models/vtransforms/base: line 38: add_depth_features=True -> False
to make the [bevfusion-det.pth](https://bevfusion.mit.edu/files/pretrained_updated/bevfusion-det.pth) work.

The result is close to the official one mAP: 68.52, NDS: 71.38.
```
Results writes to /tmp/tmp1zu28owa/results/results_nusc.json
mAP: 0.6727                              
mATE: 0.2859
mASE: 0.2559
mAOE: 0.3163
mAVE: 0.2518
mAAE: 0.1866
NDS: 0.7067
Eval time: 90.4s

Per-class results:
Object Class	AP	ATE	ASE	AOE	AVE	AAE
car	0.875	0.172	0.154	0.064	0.275	0.183
truck	0.639	0.326	0.188	0.102	0.243	0.217
bus	0.745	0.334	0.192	0.055	0.417	0.272
trailer	0.421	0.515	0.205	0.619	0.209	0.142
construction_vehicle	0.278	0.725	0.429	0.884	0.108	0.298
pedestrian	0.877	0.134	0.288	0.391	0.216	0.102
motorcycle	0.764	0.186	0.254	0.233	0.356	0.265
bicycle	0.614	0.164	0.258	0.442	0.192	0.013
traffic_cone	0.790	0.124	0.315	nan	nan	nan
barrier	0.725	0.179	0.276	0.057	nan	nan
```

closed 2023-07-28 by zhijian-liu