mit-han-lab/bevfusion #459: Use fusion model to infer on custom application

GDMG99, 2023-07-18:
Hi!
Thank you authors for your amazing work!
I have been working with BEVFusion for some time and I would like to implement it on a detection pipeline for a custom application. I have already trained and evaluated the model on different nuScenes-like datasets. This time I have a LiDAR and two cameras and I am already working on changing the data format to match the nuScenes one. 
I have been looking at the `class BEVFusion(Base3DFusionModel):` ([here](https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/models/fusion_models/bevfusion.py)), but it needs `encoder`, `fuser`, `decoder`, and `heads` dictionary inputs. Are these dictionaries the same as in the config yaml files?
From the BEVFusion class I believe I should use the forward function. Nevertheless, I cannot see how to input the weights of the model or the config file anywhere.
Could someone help me out here?
Thanks a lot in advance!

GDMG99, 2023-07-25:
Also, I have been trying to initialize the `class BevFusion(Base3DFusionModel)`. These are the dictionaries that I am using to initializes are from the config file (.yaml) of the pretrained weights of the fusion object detection model.
```
decoder:
    backbone:
      conv_cfg:
        bias: false
        type: Conv2d
      in_channels: 256
      layer_nums:
      - 5
      - 5
      layer_strides:
      - 1
      - 2
      norm_cfg:
        eps: 0.001
        momentum: 0.01
        type: BN
      out_channels:
      - 128
      - 256
      type: SECOND
    neck:
      in_channels:
      - 128
      - 256
      norm_cfg:
        eps: 0.001
        momentum: 0.01
        type: BN
      out_channels:
      - 256
      - 256
      type: SECONDFPN
      upsample_cfg:
        bias: false
        type: deconv
      upsample_strides:
      - 1
      - 2
      use_conv_for_no_stride: true
```
```
  encoders:
    camera:
      backbone:
        attn_drop_rate: 0.0
        convert_weights: true
        depths:
        - 2
        - 2
        - 6
        - 2
        drop_path_rate: 0.2
        drop_rate: 0.0
        embed_dims: 96
        init_cfg:
          checkpoint: pretrained/swint-nuimages-pretrained.pth
          type: Pretrained
        mlp_ratio: 4
        num_heads:
        - 3
        - 6
        - 12
        - 24
        out_indices:
        - 1
        - 2
        - 3
        patch_norm: true
[... 223 lines cut ...]
        gaussian_overlap: 0.1
        grid_size:
        - 1440
        - 1440
        - 41
        min_radius: 2
        out_size_factor: 8
        point_cloud_range:
        - -54.0
        - -54.0
        - -5.0
        - 54.0
        - 54.0
        - 3.0
        pos_weight: -1
        voxel_size:
        - 0.075
        - 0.075
        - 0.2
      type: TransFusionHead
 ```
In principle, the class properly loads the encoders, decoder, and fuser. However, with the heads, I get the following error:
` AttributeError: TransFusionHead: 'dict' object has no attribute 'assigner'`, yet assigner is in `heads['object']['train_cfg']`. 
Does anyone know how to properly initialize the class? Also I still do not see how to input the desired weights during the initialization.
Thank you in advance

zhijian-liu, 2024-05-04:
Adding support for a custom dataset is beyond the scope of this codebase, and unfortunately, we don't have the capacity to accommodate such customized requests.

closed 2024-05-04 by zhijian-liu