mit-han-lab/bevfusion #460: cannot import name 'feature_decorator_ext' from partially initialized module 'mmdet3d.ops.feature_decorator

MaxLeo-LMZ, 2023-07-20:
There were no errors after I compiled the program. However, when I followed the instructions given by the Evaluation, the following error occurred:

ImportError: cannot import name 'feature_decorator_ext' from partially initialized module 'mmdet3d.ops.feature_decorator' (most likely due to a circular import) (/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/ops/feature_decorator/__init__.py)
-------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code.. Per user-direction, the job has been aborted.
-------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[6176,1],0]
  Exit code:    1

I suspect that there is a problem with my compiled program, but the compilation does not indicate any problems, so I have no idea how to solve the problem.

JLIN77, 2023-07-22:
Same problem

K-Shao, 2023-07-22:
See #449

MaxLeo-LMZ, 2023-07-24:
> See #449
Thank you very much, this solution is very useful！

closed 2023-07-28 by zhijian-liu