mit-han-lab/bevfusion #461: create_data question

xiaopan999, 2023-07-20:
[image]
请问这种问题如何解决

VeeranjaneyuluToka, 2023-07-20:
You need to provide location information while creating .pkl files if you are experimenting with your dataset.

JSC1375, 2023-07-20:
> You need to provide location information while creating .pkl files if you are experimenting with your dataset.

I download nuscenes v1.0-mini and map expansion. After I run create_data.py by 
python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes
I find pkl file is named as like nuscenes_infos_train_radar.pkl and is not generated in bevfusion/data/nuscenes but bevfusion/nuscenes and finally create_data.py raise error for failing to find .pkl files. I change the name "nuscenes_infos_train_radar.pkl "  to "nuscenes_infos_train_radar.pkl"  and place them in bevfusion/data/nuscenes and rerun create_data.py, it works.

However, when I try to run 
torchpack dist-run -np 1 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map
I get the same problem KeyError: 'location'. it would be highly appreciated if you could explain how to provide location information while creating .pkl files

minje335park, 2023-07-28:
in line 192 of the nuscenes_converter.py, try to add this line 
        location = nusc.get(
            "log", nusc.get("scene", sample["scene_token"])["log_token"]
        )["location"]
In my case, it worked.

closed 2023-07-28 by zhijian-liu

quantumdotsss, 2023-08-10:
[image]
I still get the error.
I think the location should be in the info dict. Is that right?

Lewis-Lu, 2023-10-30:
> [image] I still get the error. I think the location should be in the info dict. Is that right?

Yes, `location` should be the key of the info.

Excutioner, 2023-11-21:
> > [image] I still get the error. I think the location should be in the info dict. Is that right?
> 
> Yes, `location` should be the key of the info.

I've tried this,but there is another error

Excutioner, 2023-11-21:
> > [image] I still get the error. I think the location should be in the info dict. Is that right?
> 
> Yes, `location` should be the key of the info.

I've tried this, but there is another error 
AttributeError: 'dict' object has no attribute 'in_range_bev'
How can I solve this? Thanks for help

Excutioner, 2023-11-21:
> > [image] I still get the error. I think the location should be in the info dict. Is that right?
> 
> Yes, `location` should be the key of the info.
I've tried this
[image]
but there is another error 
AttributeError: 'dict' object has no attribute 'in_range_bev'
[image]
How can I solve this?Thanks for your help

Qizhi697, 2023-12-29:
I got the same error. I wonder how to add 'location' information to the generated .pkl file.

EpicGilgamesh, 2024-01-06:
@Qizhi697 Hi, did you managed to solve this issue?