mit-han-lab/bevfusion #463: TypeError: __init__() missing 2 required positional arguments: 'doc' and 'pos'

huzhihen, 2023-07-22:
TypeError: __init__() missing 2 required positional arguments: 'doc' and 'pos'.

JunShao0104, 2023-08-16:
Same. How do you solve the problem?

YueWangTO, 2023-08-30:
i met the same problem when train bevfusion , could you please teel me how to solve it?

huzhihen, 2023-08-31:
> i met the same problem when train bevfusion , could you please teel me how to solve it?
> 
> The para sweeps_num of loadpointsfrommultisweeps function is set to 0 in the config file of the new code, whereas it was set to sweeps 9 in the original code. I think this may be causing the problem, so you can try it
> 
> update: I have tried the above methods and the problem has been solved

Thank you very much for your valuable suggestion！

I use the newest code and change sweeps_num = 0->9 at line 33 in bevfusion/configs/nuscenes/det/default.yaml to train.
The training performance is very close to the officially announced mAP of 68.52 and NDS of 71.38.
The training results are as follows：
```
Results writes to /tmp/tmpkpx7_6_a/results/results_nusc.json
mAP: 0.6805                              
mATE: 0.2860
mASE: 0.2535
mAOE: 0.3113
mAVE: 0.2557
mAAE: 0.1878
NDS: 0.7108
Eval time: 96.7s

Per-class results:
Object Class	AP	ATE	ASE	AOE	AVE	AAE
car	0.884	0.170	0.148	0.060	0.274	0.186
truck	0.636	0.321	0.181	0.087	0.247	0.221
bus	0.742	0.336	0.187	0.062	0.437	0.272
trailer	0.430	0.532	0.206	0.571	0.216	0.136
construction_vehicle	0.295	0.724	0.428	0.873	0.117	0.295
pedestrian	0.877	0.132	0.283	0.397	0.216	0.103
motorcycle	0.780	0.186	0.248	0.228	0.350	0.277
bicycle	0.639	0.163	0.255	0.470	0.188	0.012
traffic_cone	0.792	0.119	0.320	nan	nan	nan
barrier	0.729	0.176	0.277	0.054	nan	nan

```

ldtho, 2023-09-14:
Hi @huzhihen, you're using the newest code, which mean it is BevFusion-R (with extra radar information), right?

YueWangTO, 2023-09-14:
感谢您的来信，已收到

huzhihen, 2023-09-14:
> Hi @huzhihen, you're using the newest code, which mean it is BevFusion-R (with extra radar information), right?

Yes, you are right.

ldtho, 2023-09-14:
> > Hi @huzhihen, you're using the newest code, which mean it is BevFusion-R (with extra radar information), right?
> 
> Yes, you are right.

I see, do you have any results of the Lidar + Camera model only? I've been trying to reproduce that model but couldn't get the reported result. Thank you in advance :)

Gang-Chen-China, 2023-10-18:
Hi @huzhihen，do you have any results of the Radar + Camera model?

YueWangTO, 2023-10-18:
感谢您的来信，已收到

konyul, 2023-12-15:
> Thank you very much for your valuable suggestion！
>
> I use the newest code and change sweeps_num = 0->9 at line 33 in bevfusion/configs/nuscenes/det/default.yaml to train. The training performance is [...]

I also changed the sweep 0 -> 9, but the performance is still lower that the original results,
To check the difference of training detail, could you share the training log?

YueWangTO, 2023-12-15:
感谢您的来信，已收到

971022jing, 2024-01-05:
> far lower than the official mAP of 68.52 and NDS of 71.38.
> May I ask if anyone has encountered this problem？
Hello，have you solved the problem? I have the same problem that I changed the sweep 0 -> 9, but the performance is still lower that the original results.

YueWangTO, 2024-01-05:
感谢您的来信，已收到

Li-Whasaka, 2024-01-23:
I have trans. bevfusion to other projects, but mAP is very low and I don't know what the problem is. Here is my configuration file：

_base_ = [#'../_base_/datasets/nus-3d.py',
          '../_base_/default_runtime.py']

voxel_size = [0.075, 0.075, 0.2]
point_cloud_range = [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]
class_names = [
    'car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier',
    'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'
]
metainfo = dict(classes=class_names)
input_modality = dict(use_lidar=True, use_camera=True)
backend_args = None
data_config={
    'cams': ['CAM_FRONT_LEFT', 'CAM_FRONT', 'CAM_FRONT_RIGHT',
             'CAM_BACK_LEFT', 'CAM_BACK', 'CAM_BACK_RIGHT'],
    'Ncams': 6,
    'input_size': (256, 704),
    'src_size': (900, 1600),

    # Augmentation
    'resize': (-0.06, 0.11),
    'rot': (-5.4, 5.4),
    'flip': True,
    'crop_h': (0.0, 0.0),
    'resize_test':0.04,
}

device='cuda'
model = dict(
    type = 'BEVFusion',
    encoders = dict(
        camera = dict(
            backbone = dict(
                type = 'SwinTransformer',
                embed_dims = 96,
                depths = [2, 2, 6, 2],
                num_heads = [3, 6, 12, 24],
                window_size = 7,
                mlp_ratio = 4,
                qkv_bias = True,
                qk_scale = None,
                drop_rate = 0.0,
                attn_drop_rate = 0.0,
                drop_path_rate = 0.2,  
                patch_norm = True,
                out_indices = [1, 2, 3],
                with_cp = False,
                convert_weights = True,
                init_cfg=dict(
                    type='Pretrained',
                    checkpoint= 'pretrained/checkpoint/swint-nuimages-pretrained.pth' # noqa: E251
                    # 'https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth'  # noqa: E501
                    )

                # pretrained='torchvision://resnet50',
                # # pretrained='/home/lwx/Desktop/lwx/distill-bev/resnet50.pth',
                # type='ResNet',
                # depth=50,
                # num_stages=4,
                # out_indices=(1,2, 3),
                # frozen_stages=-1,
                # norm_cfg=dict(type='BN', requires_grad=True),
                # norm_eval=False,
                # with_cp=True,
                # style='pytorch'
            ),
            neck=dict(
                type='GeneralizedLSSFPN',
[... 333 lines cut ...]
# Optimizer
optimizer = dict(type='AdamW', lr=2e-4, weight_decay=0.01)
optimizer_config = dict(
    grad_clip=dict(max_norm=35,norm_type=2)
)
lr_config = dict(
    policy='step',
    warmup='linear',
    warmup_iters=500,
    warmup_ratio=0.001,
    step=[16, 22])
# Default setting for scaling LR automatically
#   - `enable` means enable scaling LR automatically
#       or not by default.
#   - `base_batch_size` = (8 GPUs) x (4 samples per GPU).
# auto_scale_lr = dict(enable=False, base_batch_size=32)

runner = dict(type='EpochBasedRunner', max_epochs=24)
evaluation = dict(interval=1)
checkpoint=dict( interval=1)
# default_hooks = dict(
#     logger=dict(type='LoggerHook', interval=50),
#     checkpoint=dict(type='CheckpointHook', interval=1))
# custom_hooks = [dict(type='DisableObjectSampleHook', disable_after_epoch=15)]
# del _base_.custom_hooks

wyf0414, 2024-03-28:
> I also changed the sweep 0 -> 9, but the performance is still lower that the original results, To check the difference of training detail, could you share the training log?

I met the same problem.
Have you solved it?

wyf0414, 2024-03-28:
> I also changed the sweep 0 -> 9, but the performance is still lower that the original results, To check the difference of training detail, could you share the training log?

@konyul 
I met the same problem.
Have you solved it?

YueWangTO, 2024-03-28:
感谢您的来信，已收到

Li-Whasaka, 2024-04-08:
> @konyul I met the same problem. Have you solved it?

Actually, I haven't solved this problem yet. Currently, I am researching other aspects. If it's resolved, I hope you can let me know

Mulbetty, 2024-07-13:
> loadpointsfrommultisweeps

Hi, I think you have already make it,I met the same problem,would you like to tell me the reason and method to solve it?

YueWangTO, 2024-07-13:
感谢您的来信，已收到

closed 2024-07-31 by zhijian-liu