mit-han-lab/bevfusion #464: Docker evaluation problem

JLIN77, 2023-07-22:
I run it by using docker env. It returns the below problems during evaluation.
Traceback (most recent call last):
  File "tools/train.py", line 15, in <module>
    from mmdet3d.datasets import build_dataset
  File "/home/bevfusion/mmdet3d/datasets/__init__.py", line 4, in <module>
    from .custom_3d import *
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 10, in <module>
    from ..core.bbox import get_box_type
  File "/home/bevfusion/mmdet3d/core/__init__.py", line 2, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/home/bevfusion/mmdet3d/core/bbox/__init__.py", line 3, in <module>
    from .iou_calculators import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/home/bevfusion/mmdet3d/core/bbox/iou_calculators/__init__.py", line 1, in <module>
    from .iou3d_calculator import (
  File "/home/bevfusion/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 5, in <module>
    from ..structures import get_box_type
  File "/home/bevfusion/mmdet3d/core/bbox/structures/__init__.py", line 1, in <module>
    from .base_box3d import BaseInstance3DBoxes
  File "/home/bevfusion/mmdet3d/core/bbox/structures/base_box3d.py", line 5, in <module>
    from mmdet3d.ops.iou3d import iou3d_cuda
  File "/home/bevfusion/mmdet3d/ops/__init__.py", line 12, in <module>
    from .feature_decorator import feature_decorator
  File "/home/bevfusion/mmdet3d/ops/feature_decorator/__init__.py", line 1, in <module>
    from .feature_decorator import feature_decorator
  File "/home/bevfusion/mmdet3d/ops/feature_decorator/feature_decorator.py", line 4, in <module>
    from mmdet3d.ops.feature_decorator import feature_decorator_ext
ImportError: cannot import name 'feature_decorator_ext' from partially initialized module 'mmdet3d.ops.feature_decorator' (most likely due to a circular import) (/home/bevfusion/mmdet3d/ops/feature_decorator/__init__.py)
--------------------------------------------------------------------------------------------------------------
Help!!  Thanks a lot!

GDMG99, 2023-07-24:
Take a look at #449

JLIN77, 2023-07-26:
First, I solved this problem by looking at #449 . Thanks a lot. But I got new problem:
[                                                  ] 0/6019, elapsed: 0s, ETA:Traceback (most recent call last):
  File "tools/test.py", line 232, in <module>
    main()
  File "tools/test.py", line 205, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/opt/conda/lib/python3.8/site-packages/mmdet/apis/test.py", line 96, in multi_gpu_test
    for i, data in enumerate(data_loader):
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/opt/conda/lib/python3.8/site-packages/torch/_utils.py", line 434, in reraise
    raise exception
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 291, in __getitem__
    return self.prepare_test_data(idx)
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 180, in prepare_test_data
    input_dict = self.get_data_info(index)
  File "/home/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 262, in get_data_info
    camera_intrinsics[:3, :3] = camera_info["cam_intrinsic"]
KeyError: 'cam_intrinsic'
How can I fix it?

VeeranjaneyuluToka, 2023-07-28:
This error is basically means your info_xx.pkl files does not have cam_intrinsic, plz check your info files.

poddar414, 2023-07-29:
I am also facing this issue. I created data again then also this problem is there. I guess there are some name mismatches. 

I tired "camera_intrinsics" instead of 'cam_intrinsic' and atleast it passed this location. Not sure why this kind of mismatch is there

closed 2024-05-04 by zhijian-liu