mit-han-lab/bevfusion #468: BEVGridTransform for non different canvas sizes

sandeepnmenon, 2023-07-25:
Thank you for the amazing work. It seems so extensible and I am able to run many experiments on this architecture.
I had some clarifying questions on the [BEVGridTransform](https://github.com/mit-han-lab/bevfusion/blob/cfc1acb42a9d6590a7cbd5a27f648aaec08febb6/mmdet3d/models/heads/segm/vanilla.py#L47) class and its parameters `grid_transform`
```yaml
grid_transform:
    input_scope: [[-51.2, 51.2, 0.8], [-51.2, 51.2, 0.8]]
    output_scope: [[-50, 50, 0.5], [-50, 50, 0.5]]
```

From what I understand the `output_scope` above generates a `200x200` size canvas. 
Does that canvas represent -50 to +50 meters with 0.5m cells?
Would it be conceptually different using `[[-50, 50, 0.5], [-50, 50, 0.5]]` vs `[[-100, 100, 1.0], [-100, 100, 1.0]]` ?
Also, how are the values for the `input_scope` determined? 
For my problem statement, I am using my custom lidar and I want to generate a canvas of size `80x200` representing 80mx200m around my ego

ccblublu, 2023-11-20:
have you solved it？ @sandeepnmenon

sandeepnmenon, 2023-11-20:
> Does that canvas represent -50 to +50 meters with 0.5m cells?
Yes

> Would it be conceptually different using [[-50, 50, 0.5], [-50, 50, 0.5]] vs [[-100, 100, 1.0], [-100, 100, 1.0]] ?
Yes they are conceptually different. Eventhough both produce same dimension output matrix, one ranges from -50m to +50m with each grid cell representing 0.5m whereas the other is -100 to 100 with 1m cells.

> Also, how are the values for the input_scope determined?
They are determined based on the lidar range of the data. It should agree with the voxelisation and filtering parameters at the data side.

> For my problem statement, I am using my custom lidar and I want to generate a canvas of size 80x200 representing 80mx200m around my ego
For this output scope can be `[[-40, 40, 1], [-100, 100, 1]]` meaning each cell would represent 1mx1m. Input scope would be similar but preferably slightly larger as shown in the example

closed 2023-11-20 by sandeepnmenon