mit-han-lab/bevfusion #472: only one GPU was occupied

wulinaaa, 2023-08-01:
When training, I run: torchpack dist-run -np 2 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth --run-dir output/bev_result/
I looked at the GPU usage and found that only one GPU was occupied, how do I make the other piece also occupied.

quantumdotsss, 2023-08-03:
> When training, I run: torchpack dist-run -np 2 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml [...]

Hi Have you solved the problem? It seems I'm using the CPU rather than the GPU...

quantumdotsss, 2023-08-07:
I'm facing the same issue now. Have you solved this?
[image]

quantumdotsss, 2023-08-07:
> I'm facing the same issue now. Have you solved this? [image]

vram, I think

wulinaaa, 2023-08-09:
> > 我现在面临同样的问题。你解决了这个问题吗？[image]
> 
> 弗拉姆，我想

I think the memory was not freed when running torchpack before. I first looked at the GPU usage and found that there was a process that kept taking up memory, so I killed it. But killing it remains. Then I'll reboot just fine.

closed 2024-05-04 by zhijian-liu