mit-han-lab/bevfusion #478: 7. RuntimeError: Given groups=1, weight of size [8, 1, 1, 1], expected input[12, 6, 256, 704] to have 1 channels, but got 6 channels instead

MaxLeo-LMZ, 2023-08-04:
**This problem occurs when I run test.py:**

 outputs = single_gpu_test(model, data_loader)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/memory_profiler.py", line 1188, in wrapper
    val = prof(func)(*args, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/memory_profiler.py", line 761, in f
    return func(*args, **kwds)
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/apis/test.py", line 13, in single_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward
    outputs = self.forward_single(
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 301, in forward_single
    feature = self.extract_camera_features(
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/models/vtransforms/base.py", line 349, in forward
    x = self.get_cam_feats(img, depth, mats_dict)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 88, in get_cam_feats
    d = self.dtransform(d)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/container.py", line 141, in forward
    input = module(input)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/conv.py", line 446, in forward
    return self._conv_forward(input, self.weight, self.bias)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/conv.py", line 442, in _conv_forward
    return F.conv2d(input, weight, bias, self.stride,
RuntimeError: Given groups=1, weight of size [8, 1, 1, 1], expected input[6, 6, 256, 704] to have 1 channels, but got 6 channels instead

**I'm using the latest code and using the nuscense dataset.**

Linkon87, 2023-08-04:
same problem @zhijian-liu

ChunyuFeng, 2023-08-05:
same problem @zhijian-liu

Ricardolyg, 2023-08-07:
i have the same problem

AlexiswaryaLucid, 2023-08-07:
I have the same problem here -latest code on nuscenes dataset

quantumdotsss, 2023-08-07:
[image]
Same here

AlexIlis, 2023-08-10:
`mmdet3d/models/vtransforms/base:` line 38: add_depth_features=True -> False works for me. Thanks @JunShao0104

Linkon87, 2023-08-11:
> `mmdet3d/models/vtransforms/base:` line 38: add_depth_features=True -> False works for me. Thanks @JunShao0104

Thank you. However, this will make val results worse , wonder is there any other solution?

Linkon87, 2023-08-11:
> > `mmdet3d/models/vtransforms/base:` line 38: add_depth_features=True -> False works for me. Thanks @JunShao0104
> 
> Thank you. However, this will make val results worse , wonder is there any other solution?

Try to use old branch may solve this , not the latest code

Ricardolyg, 2023-08-14:
> `mmdet3d/models/vtransforms/base:` line 38: add_depth_features=True -> False works for me. Thanks @JunShao0104

This works for me, but I get a new error when I run test. I feel this error has something to do with "add_depth_features=True -> False"

Ricardolyg, 2023-08-14:
> > `mmdet3d/models/vtransforms/base:` line 38: add_depth_features=True -> False works for me. Thanks @JunShao0104
> 
> This works for me, but I get a new error when I run test. I feel this error has something to do with "add_depth_features=True -> False"

Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmdet/apis/test.py", line 98, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/dell/ricardo/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward
    outputs = self.forward_single(
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/dell/ricardo/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 301, in forward_single
    feature = self.extract_camera_features(
  File "/home/dell/ricardo/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/dell/ricardo/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/dell/ricardo/bevfusion/mmdet3d/models/vtransforms/base.py", line 291, in forward
    cur_coords = torch.inverse(cur_lidar_aug_matrix[:3, :3]).matmul(
RuntimeError: cusolver error: CUSOLVER_STATUS_INTERNAL_ERROR, when calling `cusolverDnCreate(handle)`
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[60520,1],0]
  Exit code:    1

Ricardolyg, 2023-08-14:
Amazing, I took the mmdet3d/models/vtransforms/base: line 38: "add_depth_features=False" changed back to true, and then I ran test as before, and then I changed back to False, and both errors disappeared, and test ran successfully.

JunShao0104, 2023-08-14:
Hey guys, I think I have solved the problem of #455 and #478. For both evaluation and training, we should set not only the `add_depth_features=False` but `height_expand=False` at line 37 and line 38 in **bevfusion/mmdet3d/models/vtransforms/base.py**. These two flags, from my point of view, are for the Camera+Radar config. If these two flags are set to `False`, there will be no difference between the latest code and the stable old branch. In this way, the evaluation of the provided checkpoint (L+C) will be the same as the reported values.

YueWangTO, 2023-08-30:
> Hey guys, I think I have solved the problem of #455 and #478. For both evaluation and training, we should set not only the `add_depth_features=False` but `height_expand=False` at line 37 and line 38 [...]

I did as you said, set the two parameters of line 37 and 38 to false, but the training result of c+ l is still worse than the official result, may I ask you if you used the original pre-training model?

JunShao0104, 2023-08-30:
> I did as you said, set the two parameters of line 37 and 38 to false, but the training result of c+ l is still worse than the official result, may I ask you if you used the original pre-training [...]

Check the number of LiDAR sweeps in the training procedure. It should be **9** instead of 0. In this way, the re-training of the official BEVFusion LC branch will be correct.

YueWangTO, 2023-08-30:
感谢您的来信，已收到

wstzx, 2023-11-13:
Hello author, thank you so much for open sourcing your code. I tried to run your latest code using nuscenes-mini on a single GPU on 3090 and A5000, and it showed: 7.RuntimeError: Given groups=1, weight of size [8, 1, 1, 1], expected input[12, 6, 256, 704] to have 1 channels, but got 6 channels instead , so I setmmdet3d/models/vtransforms/base line37 and line38: False, but the results of the test MAP = 58, and after I used the two other old version to test, the result is only 58, far lower than the final result, is the reason I used one GPU?Here are the results of my single GPU tests in nuscenes-mini:
[image]

YueWangTO, 2023-11-13:
感谢您的来信，已收到

wstzx, 2023-11-13:
> Hello author, thank you so much for open sourcing your code. I tried to run your latest code using nuscenes-mini on a single GPU on 3090 and A5000, and it showed: 7.RuntimeError: Given groups=1, [...]
This is my run command[
[image]
](url)

gerardmartin2, 2024-04-17:
@wstzx Did you solve it?

zhijian-liu, 2024-07-31:
Thank you for your interest in our project. This repository is no longer actively maintained, so we will be closing this issue. Please refer to the amazing implementation at [MMDetection3D](https://github.com/open-mmlab/mmdetection3d/tree/main/projects/BEVFusion). Thank you again!

closed 2024-07-31 by zhijian-liu

YueWangTO, 2024-07-31:
感谢您的来信，已收到