mit-han-lab/bevfusion #479: TypeError: CyclicLrUpdaterHook: __init__() got an unexpected keyword argument 'min_lr_ratio

quantumdotsss, 2023-08-06:
Hi there,
I've run into this problem.
[image]
I check the default.yaml and there is no min_lr_ratio
but the log file has that
lr_config = dict(
    policy='cyclic',
    warmup='linear',
    warmup_iters=500,
    warmup_ratio=0.33333333,
    min_lr_ratio=0.001,
    target_ratio=5.0,
    cyclic_times=1,
    step_ratio_up=0.4)

where is the wrong part? torchpack?

Ricardolyg, 2023-08-06:
这个好像是权限问题，得把bevfusion项目放在当前用户目录下的任意文件夹

quantumdotsss, 2023-08-06:
> 这个好像是权限问题，得把bevfusion项目放在当前用户目录下的任意文件夹
you are right....

quantumdotsss, 2023-08-07:
> 这个好像是权限问题，得把bevfusion项目放在当前用户目录下的任意文件夹

NO. It's a learning ratio

Anna1ximi, 2023-08-08:
I simply delete the '**min_lr_ratio: 1.0e-3**' in **configs/nuscenes/det/centerhead/lssfpn/default.yaml** and it works.

quantumdotsss, 2023-08-09:
> @quantumdotsss I simply delete the '**min_lr_ratio: 1.0e-3**' in **configs/nuscenes/det/centerhead/lssfpn/default.yaml** and it works.

yeap!

closed 2023-08-09 by quantumdotsss