mit-han-lab/bevfusion #480: FileNotFoundError: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

Ricardolyg, 2023-08-06:
运行python  tools/create_data.py  nuscenes --root-path ./data/nuscenes/ --version v1.0-mini --out-dir data/nuscenes/ --extra-tag nuscenes时报错：
Loading NuScenes tables for version v1.0-mini...
23 category,
8 attribute,
4 visibility,
911 instance,
12 sensor,
120 calibrated_sensor,
31206 ego_pose,
8 log,
10 scene,
404 sample,
31206 sample_data,
18538 sample_annotation,
4 map,
Done loading in 0.580 seconds.
======
Reverse indexing ...
Done reverse indexing in 0.1 seconds.
======
total scene num: 10
exist scene num: 10
train scene: 8, val scene: 2
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] 404/404, 1.1 task/s, elapsed: 357s, ETA:     0s
nuscenes
train sample: 323, val sample: 81
Create GT Database of NuScenesDataset
Traceback (most recent call last):
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/dell/ricardo/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/home/dell/ricardo/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/home/dell/ricardo/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 202, in load_annotations
    data = mmcv.load(ann_file)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/fileio/io.py", line 60, in load
    with BytesIO(file_client.get(file)) as f:
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 992, in get
    return self.client.get(filepath)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 517, in get
    with open(filepath, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/create_data.py", line 122, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 45, in nuscenes_data_prep
    create_groundtruth_database(
  File "/home/dell/ricardo/bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database
    dataset = build_dataset(dataset_cfg)
  File "/home/dell/ricardo/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/home/dell/anaconda3_ricardo/envs/bevfusion06/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: 'data/nuscenes//nuscenes_infos_train.pkl'

Ricardolyg, 2023-08-06:
各位大佬这个如何解决？

Ricardolyg, 2023-08-06:
运行create.py不就是为了生成pkl文件吗，然后报错说没有，不懂，请求大佬们帮忙

MaxLeo-LMZ, 2023-08-07:
You can check to see if there is an extra nuscense folder in the bevfusion directory, and the folder contains nuscense_infos_train_radar.pkl and nuscense_infos_val_radar.pkl. If the file exists, then modify line 102 of bevfusion->tools->data_converter->nuscenes_converter.py

```
info_path = osp.join(info_prefix,
                     '{}_infos_train_radar.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(info_prefix,
                         '{}_infos_val_radar.pkl'.format(info_prefix))
```
Be changed to
```
info_path = osp.join(info_prefix,
                     '{}_infos_train.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(info_prefix,
                         '{}_infos_val.pkl'.format(info_prefix))
```
Then run create_data.py again and you will find nuscense_infos_train.pkl and nuscense_infos_val.pkl in the bevfusion->nusences folder, just copy the two files to your nuscenes dataset directory.

Ricardolyg, 2023-08-07:
Thank you very much indeed. According to your method, the problem has been solved

Zhanfury, 2023-12-06:
It seems that this error happened because the authers renamed the output 'pkl‘ files after adding radar data.
Just change“info_path = osp.join(info_prefix,
                     '{}_infos_train_radar.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(info_prefix,
                         '{}_infos_val_radar.pkl'.format(info_prefix)) ”
to the following code:       
"info_path = osp.join('data',info_prefix,
                             '{}_infos_train.pkl'.format(info_prefix))
        mmcv.dump(data, info_path)
        data['infos'] = val_nusc_infos
        info_val_path = osp.join('data',info_prefix,
                                 '{}_infos_val.pkl'.format(info_prefix))
        mmcv.dump(data, info_val_path)"
and your nuscense_infos_train.pkl and nuscense_infos_val.pkl will be directly in data/nuscenes/.

CaveSpiderLZJ, 2024-01-18:
In addition to MaxLeo-LMZ's comments, you can just change:
```
info_path = osp.join(info_prefix,
                     '{}_infos_train_radar.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(info_prefix,
                         '{}_infos_val_radar.pkl'.format(info_prefix))

```
to
```
info_path = osp.join(root_path,
                     '{}_infos_train.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(root_path,
                         '{}_infos_val.pkl'.format(info_prefix))
```
It seems the author used `info_prefix` twice.

hitbuyi, 2024-04-17:
> In addition to MaxLeo-LMZ's comments, you can just change:
>
> ```
> info_path = osp.join(info_prefix,
>                      '{}_infos_train_radar.pkl'.format(info_prefix))
> mmcv.dump(data, info_path) [...]

This is final solution, use root_path  to replace info_prefix

closed 2024-05-04 by zhijian-liu