mit-han-lab/bevfusion #484: can't use np>1

Anna1ximi, 2023-08-08:
**When I run the command:** torchpack dist-run -np 4 python tools/train.py \
	configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml \
	--model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth \
	--load_from pretrained/lidar-only-det.pth \
	--run-dir output/bev_result/
**I get this problem:**
[image]
**And when I want to train the camera-only variant for object detection, I get this:**
[image]
[image]
**Does anyone know how to fix it？Thanks！**

quantumdotsss, 2023-08-09:
I think you need to provide more error info

VanderBieu, 2023-08-11:
the number of np must match the number of gpus.

Anna1ximi, 2023-08-11:
That's all the information, only a few lines. @quantumdotsss

Anna1ximi, 2023-08-11:
I know,  I adhere strictly to this point,  but it always goes wrong. @VanderBieu

enhuima, 2023-10-19:
same question.

Anna1ximi, 2023-10-19:
> same question.

I started with the environment configuration all over again and it worked.

ChirsHuanhuan, 2023-10-19:
[image]
same question when I run the test.py 
error: address already use

huzhihen, 2023-11-05:
> [image] same question when I run the test.py error: address already use

Hello, I have also encountered this problem. Have you resolved it?

closed 2024-05-04 by zhijian-liu