mit-han-lab/bevfusion #486: How much GPU memory is needed to train BEVFusion's model

MaxLeo-LMZ, 2023-08-10:
I have a problem：
    train_model(
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/apis/train.py", line 133, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 75, in train_step
    return self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/fusion_models/bevfusion.py", line 230, in forward_single
    feature = self.extract_camera_features(
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/fusion_models/bevfusion.py", line 114, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/vtransforms/base.py", line 293, in forward
    x = self.bev_pool(geom, x)
  File "/home/mingzhong/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/mingzhong/PycharmProjects/bevfusion_noradar/mmdet3d/models/vtransforms/base.py", line 155, in bev_pool
    x = x[kept]
**RuntimeError: CUDA out of memory. Tried to allocate 532.00 MiB (GPU 0; 7.78 GiB total capacity; 3.79 GiB already allocated; 526.19 MiB free; 4.29 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF**

My GPU is currently **NVIDIA RTX3070 8G**
Should I replace a GPU, and at least what type of GPU should I replace it with？
Hope to get your suggestions

quantumdotsss, 2023-08-10:
3080 at least

VanderBieu, 2023-08-11:
to my knowledge if you want to train the fusion model, then the requirement of VRAM should be at least 24GB for batchsize=1.

wyf0414, 2023-11-28:
As for me, 4090.
It needs about 20000 MB to train.
sample_per_gpu=1

flstahl, 2024-01-17:
Hey folks,
Can anyone help me understand where this memory consumption comes from?

I am currently working with BEVFusion and having serious trouble with the GPU memory. I am having trouble understanding why this model needs so much memory for training. Given that the model size is ~150M params and let's assume an AdamW optimizer.

I found that you would need:

    4 bytes * number of parameters for fp32 training
    8 bytes * number of parameters for normal AdamW (maintains 2 states)

This comes to just shy of 2GB for the Model. Let's say we have a sample data size of 100 MB, I am having trouble understanding how we end up maxing out 24GB GPUs here.

Is there something major I am missing?

Any help would be highly appreciated in helping me getting my head around this! Thanks a lot

josh-wende, 2024-01-25:
@wyf0414 thanks for sharing. How long does it take you to train the model on the number of 4090s that you use?

esh64, 2024-02-05:
@josh-wende 

I'm using a single RTX4090 and it took about 3 days to train the Camera+Lidar detection model with some modifications in the learning rate.

josh-wende, 2024-02-05:
@esh64 thank you, that is very helpful! Are you still able to replicate the quality of the published results with these modifications?

Also, I noticed that the authors say that for training a fusion camera+lidar model, one should first train a lidar-only model, then use that as the basis for training the fusion model. Does your time estimate include all of that, or just the last step?

esh64, 2024-02-06:
@josh-wende 

I have not performed training for lidar-only. I used the author's pre-trained model available for download. The result was very close to that reported by the author

LX0912R, 2024-02-22:
> @josh-wende
> 
> I'm using a single RTX4090 and it took about 3 days to train the Camera+Lidar detection model with some modifications in the learning rate.

Hello, are you using cuda11.8 and pytorch2.0+ when training on RTX4090?

closed 2024-05-04 by zhijian-liu