mit-han-lab/bevfusion #488: TypeError: 'DataContainer' object is not iterable

Anna1ximi, 2023-08-13:
**When I run the command for visualization, I get this error:**

load checkpoint from local path: pretrained/bevfusion-det.pth
  0% 0/6019 [00:02<?, ?it/s]
Traceback (most recent call last):
  File "tools/visualize.py", line 162, in <module>
    main()
  File "tools/visualize.py", line 84, in main
    outputs = model(**data)
  File "/root/miniconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/root/autodl-tmp/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward
    outputs = self.forward_single(
  File "/root/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/root/autodl-tmp/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 319, in forward_single
    feature = self.extract_features(points, sensor)
  File "/root/autodl-tmp/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 152, in extract_features
    feats, coords, sizes = self.voxelize(x, sensor)
  File "/root/miniconda3/envs/bev/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "/root/miniconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func
    return old_func(*args, **kwargs)
  File "/root/autodl-tmp/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 173, in voxelize
    for k, res in enumerate(points):
TypeError: 'DataContainer' object is not iterable

**Does anyone know how to fix it? Thanks very much!**

Anna1ximi, 2023-08-15:
I just changed tools/visualize.py to single-card mode, and the rest remains unchanged. Why does this error occur? @kentang-mit 
The command I used is: torchpack dist-run -np 1 python tools/visualize.py \
	configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml \
	--checkpoint=pretrained/bevfusion-det.pth \
	--out-dir vision/pred_result \
	--mode pred \
	--bbox-score 0.2

sym330, 2024-01-04:
I got the same error, did you fix that?

chuliuT, 2024-07-26:
@sym330 @Anna1ximi ,use the MMDataParallel warp the model,just like this below code:
from mmcv.parallel import MMDataParallel,MMDistributedDataParallel
    # build the model and load checkpoint
    if args.mode == "pred":
        model = build_model(cfg.model)
        fp16_cfg = cfg.get("fp16", None)
        if fp16_cfg is not None:
            wrap_fp16_model(model)
        load_checkpoint(model, args.checkpoint, map_location="cpu")
        model = MMDataParallel(model, device_ids=[0])
        model.eval()

closed 2024-07-31 by zhijian-liu