mit-han-lab/bevfusion #491: 'dict' object has no attribute 'rotate'

quantumdotsss, 2023-08-15:
[image]
Hi guys,
I would like to know how the radar data is used.
[image]
from the figure of the paper, I didn't see the usage of ladar
[image]
So I think I can just comment those lines.
Am I right.
This happens when I was running 
torchpack dist-run -np 1 python tools/train.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth

Charlie0804, 2024-03-09:
I'm also having the same issue, would like to ask if you solved it

zhijian-liu, 2024-05-04:
Feel free to comment out the code related to radar if you are only working on lidar-only, camera-only, or camera-lidar fusion models.

closed 2024-05-04 by zhijian-liu