mit-han-lab/bevfusion #492: ModuleNotFoundError: No module named 'flash_attn'

liwenxiang12, 2023-08-16:
When I do a test ，there is an error here, but I can't find the reason.
**test command:**
_torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml pretrained/swint-nuimages-pretrained.pth --eval bbox --out box.pkl_
**and I got this error:**
Traceback (most recent call last):
  File "tools/test.py", line 16, in <module>
    from mmdet3d.models import build_model
  File "/home/lwx/bev/bevfusion/mmdet3d/models/__init__.py", line 1, in <module>
    from .backbones import *
  File "/home/lwx/bev/bevfusion/mmdet3d/models/backbones/__init__.py", line 9, in <module>
    from .radar_encoder import *
  File "/home/lwx/bev/bevfusion/mmdet3d/models/backbones/radar_encoder.py", line 18, in <module>
    from flash_attn.flash_attention import FlashMHA
ModuleNotFoundError: No module named 'flash_attn'
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[64587,1],0]
  Exit code:    1
--------------------------------------------------------------------------

**I tried to install flash_attn module, and it failed.**
Because the cuda and pytorch version that flash_atten requires  exceed the required version for bevfusion.

quantumdotsss, 2023-08-16:
pip install flash-attn==0.2.0
easy fix

liwenxiang12, 2023-08-17:
> pip install flash-attn==0.2.0 easy fix

Thank you for your reply, but I have tried it before and it doesn't work

quantumdotsss, 2023-08-17:
install and then re-compile

python setup.py develop

wyf0414, 2023-11-28:
Just comment the code `from flash_attn.flash_attention import FlashMHA`
and the code `from .radar_encoder import *`.
Because they are used for radar.
Hope this can help you.

hsingyu-chou, 2024-03-28:
Try this.

/bevfusion/mmdet3d/models/backbones/radar_encoder.py :

from flash_attn.flash_attention import FlashMHA
→ from flash_attn.modules.mha import MHA

I think this error is caused by the difference between the version 1.0.9 and 2.0.0 of flash-attention.

closed 2024-05-04 by zhijian-liu