mit-han-lab/bevfusion #497: Evalution RuntimeError: cusolver error: CUSOLVER_STATUS_INTERNAL_ERROR, when calling `cusolverDnCreate(handle)`

cloudlockheart, 2023-08-22:
Hi guys

I meet a problem when I do fusion-det or fusion-seg evalution, the error log shows as below:
```
{
[                                                  ] 0/6019, elapsed: 0s, ETA:Traceback (most recent call last):
  File "tools/test.py", line 249, in <module>
    main()
  File "tools/test.py", line 212, in main
    outputs = single_gpu_test(model, data_loader)
  File "/home/bevfusion/mmdet3d/apis/test.py", line 12, in single_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward
    outputs = self.forward_single(
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 301, in forward_single
    feature = self.extract_camera_features(
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/vtransforms/base.py", line 291, in forward
    cur_coords = torch.inverse(cur_lidar_aug_matrix[:3, :3]).matmul(
RuntimeError: cusolver error: CUSOLVER_STATUS_INTERNAL_ERROR, when calling `cusolverDnCreate(handle)`
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[2148,1],2]
  Exit code:    1
}
```

I use single 3080 GPU for the evalution, and have done the data preparation, is anyone know how to solve this error?

Thanks

Lesliewsq, 2023-11-16:
Same error，have you solved it？

hankeceli, 2023-12-19:
same error as well, have you found a solution?

Wangquans, 2024-02-28:
> same error as well, have you found a solution?

Charlie0804, 2024-03-09:
same error as well, have you found a solution?

Wangquans, 2024-04-07:
i still have no solutions.I am trying to change  my cuda version from 11.1
to 11.3 to solve the problem

closed 2024-07-31 by zhijian-liu