mit-han-lab/bevfusion #502: Wrong Visualization on Training Set

NoOneUST, 2023-09-06:
I can generate correct visualizations on validation and testing sets, but the results are wrong on training sets.

[image]
[image]

These two images are the gt and pred results on the training set. I have confirmed that this error impacts all images in the training set.

Anna1ximi, 2023-09-06:
@NoOneUST  Hey! I want to know how you modified visualize.py for visualization. I have been stuck on this question for days. Thanks!

sangjinpark97, 2023-09-17:
I'm not author or contributor of this project.

In my opinion, it seems to be related to data augmentation. Training data is processed by specific pipeline which contains augmentation.

closed 2024-05-04 by zhijian-liu