mit-han-lab/bevfusion #503: compilation terminated. error: command '/usr/bin/gcc' failed with exit code 1

nican2018, 2023-09-07:
Hello 

Please, I followed all step but the setup gives the following error

creating build/lib.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d
g++ -pthread -shared -B /opt/conda/envs/bevfusion/compiler_compat -L/opt/conda/envs/bevfusion/lib -Wl,-rpath=/opt/conda/envs/bevfusion/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d/src/points_in_boxes_cpu.o build/temp.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d/src/points_in_boxes_cuda.o build/temp.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d/src/roiaware_pool3d.o build/temp.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d/src/roiaware_pool3d_kernel.o -L/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/lib -L/usr/local/cuda/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda -o build/lib.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d/roiaware_pool3d_ext.cpython-38-x86_64-linux-gnu.so
creating build/temp.linux-x86_64-3.8/mmdet3d/ops/ball_query
creating build/temp.linux-x86_64-3.8/mmdet3d/ops/ball_query/src
gcc -pthread -B /opt/conda/envs/bevfusion/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DWITH_CUDA -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/TH -I/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/opt/conda/envs/bevfusion/include/python3.8 -c mmdet3d/ops/ball_query/src/ball_query.cpp -o build/temp.linux-x86_64-3.8/mmdet3d/ops/ball_query/src/ball_query.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=ball_query_ext -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
mmdet3d/ops/ball_query/src/ball_query.cpp:4:10: fatal error: THC/THC.h: No such file or directory
 #include <THC/THC.h>
          ^~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/gcc' failed with exit code 1

hsingyu-chou, 2023-10-02:
Try this.

1. \bevfusion\mmdet3d\ops\ball_query\src\ball_query.cpp
2. \bevfusion\mmdet3d\ops\knn\src\knn.cpp
3. \bevfusion\mmdet3d\ops\group_points\src\group_points.cpp
4. \bevfusion\mmdet3d\ops\interpolate\src\interpolate.cpp
5. \bevfusion\mmdet3d\ops\furthest_point_sample\src\furthest_point_sample.cpp
6. \bevfusion\mmdet3d\ops\gather_points\src\gather_points.cpp

#include <THE/THC.h>
extern THCState *state;
cudaStream_t stream = THCState_getCurrentStream(state);

Replace with →
#include <ATen/cuda/CUDAContext.h>
#include <ATen/cuda/CUDAEvent.h>
cudaStream_t stream = at::cuda::getCurrentCUDAStream();

ranai-srivastav, 2023-11-02:
@nican2018 did the above fix work for you?
@hsingyu-chou I did this but had a different issue

I'm currently running Ubuntu 22.04, with pytorch version 2.1.0+cu121. 
Here are the details about nvcc and cuda
```
nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Fri_Sep__8_19:17:24_PDT_2023
Cuda compilation tools, release 12.3, V12.3.52
Build cuda_12.3.r12.3/compiler.33281558_0

```
Error stacktrace
```

Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/1] c++ -MMD -MF /home/ranai/software/bevfusion/build/temp.linux-x86_64-3.8/mmdet3d/ops/group_points/src/group_points.o.d -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DWITH_CUDA -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/usr/include/python3.8 -c -c /home/ranai/software/bevfusion/mmdet3d/ops/group_points/src/group_points.cpp -o /home/ranai/software/bevfusion/build/temp.linux-x86_64-3.8/mmdet3d/ops/group_points/src/group_points.o -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=group_points_ext -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
FAILED: /home/ranai/software/bevfusion/build/temp.linux-x86_64-3.8/mmdet3d/ops/group_points/src/group_points.o 
c++ -MMD -MF /home/ranai/software/bevfusion/build/temp.linux-x86_64-3.8/mmdet3d/ops/group_points/src/group_points.o.d -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DWITH_CUDA -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/TH -I/home/ranai/.local/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/include -I/home/ranai/.local/lib/python3.8/site-packages/torch/in [... cut ...]
  File "/home/ranai/.local/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 686, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/home/ranai/.local/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1774, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/ranai/.local/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 2116, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

```

GaoPeng97, 2023-11-08:
> @nican2018 did the above fix work for you? @hsingyu-chou I did this but had a different issue
>
> I'm currently running Ubuntu 22.04, with pytorch version 2.1.0+cu121. Here are the details about nvcc [...]

same issue with you ,have you ever solved this ?

kbar2222, 2024-06-26:
> same issue with you ,have you ever solved this ?

> same issue with you ,have you ever solved this ?

>

same issue have you any solution plz ?

kbar2222, 2024-06-26:
same issue have you any solution plz ?

closed 2024-07-31 by zhijian-liu