mit-han-lab/bevfusion #504: ImportError: cannot import name 'feature_decorator_ext' from partially initialized module 'mmdet3d.ops.feature_decorator' (most likely due to a circular import) (/data/bevfusion/mmdet3d/ops/feature_decorator/__init__.py)

wangjunquan911, 2023-09-07:
Hellow,
The following errror occured when I ran the train.py

wangjunquan911, 2023-09-07:
root@6cadeb67f974:/data/bevfusion# sh Meg_dataset/bash_runner/train.sh
Traceback (most recent call last):
  File "Meg_dataset/tools/train.py", line 15, in <module>
    from mmdet3d.datasets import build_dataset
  File "/data/bevfusion/mmdet3d/datasets/__init__.py", line 4, in <module>
    from .custom_3d import *
  File "/data/bevfusion/mmdet3d/datasets/custom_3d.py", line 10, in <module>
    from ..core.bbox import get_box_type
  File "/data/bevfusion/mmdet3d/core/__init__.py", line 2, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/data/bevfusion/mmdet3d/core/bbox/__init__.py", line 3, in <module>
    from .iou_calculators import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/data/bevfusion/mmdet3d/core/bbox/iou_calculators/__init__.py", line 1, in <module>
    from .iou3d_calculator import (
  File "/data/bevfusion/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py", line 5, in <module>
    from ..structures import get_box_type
  File "/data/bevfusion/mmdet3d/core/bbox/structures/__init__.py", line 1, in <module>
    from .base_box3d import BaseInstance3DBoxes
  File "/data/bevfusion/mmdet3d/core/bbox/structures/base_box3d.py", line 5, in <module>
    from mmdet3d.ops.iou3d import iou3d_cuda
  File "/data/bevfusion/mmdet3d/ops/__init__.py", line 12, in <module>
    from .feature_decorator import feature_decorator
  File "/data/bevfusion/mmdet3d/ops/feature_decorator/__init__.py", line 1, in <module>
    from .feature_decorator import feature_decorator
  File "/data/bevfusion/mmdet3d/ops/feature_decorator/feature_decorator.py", line 4, in <module>
    from . import feature_decorator_ext
ImportError: cannot import name 'feature_decorator_ext' from partially initialized module 'mmdet3d.ops.feature_decorator' (most likely due to a circular import) (/data/bevfusion/mmdet3d/ops/feature_decorator/__init__.py)

lymcs, 2023-09-20:
I have encountered the same problem as you. Have you resolved it?? That great god can help with one hand

hsingyu-chou, 2023-10-02:
I found this solution from other website.
It works but I don't know why.
You could try this.

1. mmdet3d/ops/feature_decorator/feature_decorator.py : 
   Change “from mmdet3d.ops.feature_decorator import feature_decorator_ext” to “from . import feature_decorator_ext”.
2. setup.py :
   Add the below.
               make_cuda_ext(
                name="feature_decorator_ext",
                module="mmdet3d.ops.feature_decorator",
                sources=["src/feature_decorator.cpp"],
                sources_cuda=["src/feature_decorator_cuda.cu"],
            )
3. mmdet3d\ops\feature_decorator\src\feature_decorator.cpp :
   Delete the below.
   static auto registry =
    torch::RegisterOperators("feature_decorator_ext::feature_decorator_forward", &feature_decorator_forward);

qiliang72, 2023-10-04:
> I found this solution from other website. It works but I don't know why. You could try this.
>
> 1. mmdet3d/ops/feature_decorator/feature_decorator.py :
>    Change “from mmdet3d.ops.feature_decorator [...]

Thanks, it works for me!

It seems that feature_decorator is not actually used in the train/test process, so just comment out all feature_decorator import might also be a good solution.

nlitz88, 2024-02-17:
Kevin's answer [here](https://github.com/mit-han-lab/bevfusion/issues/449#issuecomment-1646598119) worked as a quick solution for me:

> Sorry for the delay! This behavior is due to a bug in the feature decorator, which is a dummy OP for the ONNX exporting, which is not actually supported in the codebase currently. I have a PR (https://github.com/mit-han-lab/bevfusion/pull/466) that will fix the issue, but it should also simply work to (as @For24 mentions) comment out the feature decorator import in mmdet3d/ops/init.py (however, you should also comment lines 15 and 177-178 in radar_encoder.py).

closed 2024-05-04 by zhijian-liu

kxxbeomjun, 2024-05-13:
Thanks a lot. It did work to me

> I found this solution from other website. It works but I don't know why. You could try this.
>
> 1. mmdet3d/ops/feature_decorator/feature_decorator.py :
>    Change “from mmdet3d.ops.feature_decorator [...]

Thanks a lot. It did work to me