mit-han-lab/bevfusion #507: Problems when replicating the result using nuscenes-mini

JohnFuguiWang, 2023-09-08:
## Problem description
I tried to train the model using nuscenes-mini dataset, and found the result statistics were quite low (around 0.45). Another weird thing was that the loss declined gradually, but the object map stayed at the same level of 0.45 from the very beginning to the end, and it even went downward with training going on. 
By the way, I've checked issue #463 and modified the sweeps_num as 9, and the result above was already the result after this modification. 
So did anyone also use nusenes-mini and how was your result? And why did the map figure stay at the same level during training? Many thanks for your help!  
## To replicate the problem
I use nuscenes-mini dataset, set the max_epochs as 15, both samples_per_gpu and workers_per_gpu as 2, and use the official order of detect: "torchpack dist-run -np 2 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth"
## More details
I can also attach some visualization results and the complete config.yaml here if you want to see more details.
[image]
[image]
[image]
/home/wcx/My_Research/My_Code/BEVFusion-mit/configs.yaml

JohnFuguiWang, 2023-09-08:
Here is the content of my config.yaml if you wanna have a look at:
augment2d:
  gridmask:
    fixed_prob: true
    prob: 0.0
  resize:
  - - 0.38
    - 0.55
  - - 0.48
    - 0.48
  rotate:
  - -5.4
  - 5.4
augment3d:
  rotate:
  - -0.78539816
  - 0.78539816
  scale:
  - 0.9
  - 1.1
  translate: 0.5
checkpoint_config:
  interval: 1
  max_keep_ckpts: 1
cudnn_benchmark: false
data:
  samples_per_gpu: 2
  test:
    ann_file: data/nuscenes/nuscenes_infos_val.pkl
    box_type_3d: LiDAR
    dataset_root: data/nuscenes/
    map_classes:
    - drivable_area
    - ped_crossing
    - walkway
    - stop_line
    - carpark_area
    - divider
    modality:
      use_camera: true
      use_external: false
      use_lidar: true
      use_map: false
      use_radar: false
    object_classes:
    - car
    - truck
    - construction_vehicle
    - bus
    - trailer
    - barrier
    - motorcycle
    - bicycle
    - pedestrian
    - traffic_cone
    pipeline:
    - to_float32: true
      type: LoadMultiViewImageFromFiles
    - coord_type: LIDAR
      load_augmented: null
      load_dim: 5
      reduce_beams: 32
      type: LoadPointsFromFile
      use_dim: 5
    - load_augmented: null
      load_dim: 5
      pad_empty_sweeps: true
      reduce_beams: 32
      remove_close: true
      sweeps_num: 9
[... 1499 lines cut ...]
  - traffic_cone
  type: DefaultFormatBundle3D
- keys:
  - img
  - points
  - radar
  - gt_bboxes_3d
  - gt_labels_3d
  meta_keys:
  - camera_intrinsics
  - camera2ego
  - lidar2ego
  - lidar2camera
  - lidar2image
  - camera2lidar
  - img_aug_matrix
  - lidar_aug_matrix
  type: Collect3D
- keyframe_only: true
  type: GTDepth
use_dim: 5
voxel_size:
- 0.075
- 0.075
- 0.2

Tamako668, 2023-10-19:
Hello, have you solved this problem?  I encountered the same problem.

JohnFuguiWang, 2023-10-21:
> Hello, have you solved this problem? I encountered the same problem.

Honestly speaking, I still don't know the reason of this problem now. I reinstalled the conda environment following this [blog](https://blog.csdn.net/m0_46556474/article/details/130561644) , and ran the model again using nuscenes-mini. The result map was around 0.47 and stayed stable. I think maybe this result is normal.
You can try to reinstall your conda environment too and run the model again. If it's possible you can notify me of your new result, and we can check if our results are normal. Thanks in advance.

closed 2024-05-04 by zhijian-liu