mit-han-lab/bevfusion #508: How to convert a trained model to onnx

YueSun0609, 2023-09-12:
Thank you for your kind release of the codes, I have successfully applied it to my own data and got good performace. But may I ask how to export a trained model to onnx format? In tools/export.py, you gave an example but missing "forward_test" function. I tried to write one, but when I run torch.onnx.export() function, it meets an error like "RuntimeError: Tracer cannot infer type of [{'boxes_3d': LiDARInstance3DBoxes", could you give me some clues on how to solve this issue? Thank you so much!

ueke616, 2024-02-05:
I got the same error, can anyone give some solution? thanks

zhijian-liu, 2024-05-04:
We currently do not support exporting the model to ONNX, but it is under consideration for future updates.

closed 2024-05-04 by zhijian-liu