mit-han-lab/bevfusion #512: I am running the official training and have encountered the following error, which has caused me to be unable to train properly. Can anyone help me?

lymcs, 2023-09-24:
bevfusion) wsd2@wsd2:~/bevfusion$ torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth
Invalid MIT-MAGIC-COOKIE-1 keyTraceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 51, in main
    logger.info(f"Config:\n{cfg.pretty_text}")
  File "/home/wsd2/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/utils/config.py", line 496, in pretty_text
    text, _ = FormatCode(text, style_config=yapf_style, verify=False)
TypeError: FormatCode() got an unexpected keyword argument 'verify'
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[36386,1],0]
  Exit code:    1

hsingyu-chou, 2023-10-02:
Try this.
!pip install yapf==0.40.1

closed 2024-05-03 by zhijian-liu

Ange1ika, 2024-05-08:
[image]
After installation, there was still an error...