mit-han-lab/bevfusion #513: location=info["location"], KeyError: 'location'

lymcs, 2023-09-25:
I encountered this issue while running the following code

torchpack dist-run -np 1 -v python tools/test.py configs/nuscenes/det/transfusion/secfpn/lidar/default.yaml pretrained/lidar-only-det.pth --eval bbox^C

Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/home/wsd/bevfusion/mmdet3d/apis/train.py", line 126, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/home/wsd/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/wsd/bevfusion/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/home/wsd/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 47, in train
    for i, data_batch in enumerate(self.data_loader):
  File "/home/wsd/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/home/wsd/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 561, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "/home/wsd/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/wsd/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/wsd/bevfusion/mmdet3d/datasets/dataset_wrappers.py", line 68, in __getitem__
    return self.dataset[ori_idx]
  File "/home/wsd/bevfusion/mmdet3d/datasets/custom_3d.py", line 293, in __getitem__
    data = self.prepare_train_data(idx)
  File "/home/wsd/bevfusion/mmdet3d/datasets/custom_3d.py", line 160, in prepare_train_data
    input_dict = self.get_data_info(index)
  File "/home/wsd/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 218, in get_data_info
    location=info["location"],
KeyError: 'location'
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[42112,1],0]
  Exit code:    1

YHao29, 2023-11-22:
I met the same issue. Are there any solutions?

J4nekT, 2023-12-02:
I am experiencing the same error. Have you already found a solution for this?

Qizhi697, 2024-01-01:
I met the same issue. Are there any solutions?

zhijian-liu, 2024-05-04:
Please refer to https://github.com/mit-han-lab/bevfusion/issues/514.

closed 2024-05-04 by zhijian-liu