mit-han-lab/bevfusion #517: KeyError: 'radar'`

chg0901, 2023-10-02:
I have an error when I try to use your code.

` File "/home/user/Documents/GitHub/bevfusion/mmdet3d/datasets/pipelines/loading.py", line 728, in __call__
    radars_dict = results['radar']
KeyError: 'radar'`

And I found a simiar iusse in another [project](https://github.com/Tsinghua-MARS-Lab/futr3d/issues/42)

Is it because I put the mmdetection3d github folder into the project and install it before I run `python setup.py develop` so the mmdetection3d could affect the result?

The following are the outputs after I run `python setup.py develop`
```
Removing mmdet3d 1.0.0rc0 from easy-install.pth file
Adding mmdet3d 0.0.0 to easy-install.pth file
detected new path '/home/cine/Documents/GitHub/bevfusion/mmdetection3d'

Installed /home/cine/Documents/GitHub/bevfusion
Processing dependencies for mmdet3d==0.0.0
Finished processing dependencies for mmdet3d==0.0.0
```
did you know how to solve this problem? Thank you very much for your help and reply~

chg0901, 2023-10-02:
Here I want to share what I have done with nuscene mini dataset and this code (even it is with some Chinese, I help this could help people want to use this code)

```
conda create --name fusionbev python=3.8 -y

pip install opencv-python
pip install pyquaternion
pip install nuscenes-devkit
pip install lyft-dataset-sdk

conda install pytorch==1.10.1 torchvision==0.11.2 torchaudio==0.10.1 cudatoolkit=11.3 -c pytorch -c conda-forge

#下载openmpi 见最下面
pip install mpi4py==3.0.3
pip install torchpack
pip install tqdm
pip install Pillow==8.4.0

#  mmcv = 1.4.0  mmdetection = 2.20.
# https://mmdetection.readthedocs.io/en/latest/get_started.html
pip install -U openmim        # 安装 mim
mim install mmengine          # 安装基础库 mmengine

# https://mmdetection.readthedocs.io/zh_CN/v2.21.0/get_started.html
### MMCV/MMDet/MMDet3D 的版本对应_csuzhaoqinghui的博客-CSDN博客
#### https://blog.csdn.net/csuzhaoqinghui/article/details/130984373

MMDetection3D 	MMDetection 	        MMSegmentation 	        MMCV
v1.0.0rc0	mmdet>=2.19.0, <=3.0.0	mmseg>=0.20.0, <=1.0.0	mmcv-full>=1.3.17, <=1.5.0

mim install "mmcv-full==1.4.0"
pip install mmdet==2.20.0

git clone https://github.com/open-mmlab/mmsegmentation.git
cd mmsegmentation
git checkout v0.20.0  # switch to v0.20.0 branch
pip install -e .  # or "python setup.py develop"

cd ..

git clone https://github.com/open-mmlab/mmdetection3d.git
cd mmdetection3d
git checkout v1.0.0rc0  # switch to v0.20.0 branch
pip install -e .  # or "python setup.py develop"

## 创建pkl文件
# https://mmdetection3d.readthedocs.io/zh_CN/dev-1.x/advanced_guides/datasets/nuscenes.html

# cd到mmdetection3d文件夹，注意root path多位置和version的设置
# ./tools/create_data.py  line 120
#     elif args.dataset == "nuscenes" and args.version == "v1.0-mini":

python ./tools/create_data.py nuscenes --root-path ../data/nuscenes --out-dir ../data/nuscenes --extra-tag nuscenes --version v1.0-mini

## mmdet3d 1.0.0rc0 requires numba==0.53.0, but you have numba 0.58.0 which is incompatible.
pip install numba==0.53.0
pip install numpy==1.23.4

# 执行完，增加文件
nuscenes_dbinfos_train.pkl
nuscenes_gt_database
nuscenes_infos_train_mono3d.coco.json
nuscenes_infos_train.pkl
nuscenes_infos_val_mono3d.coco.json
nuscenes_infos_val.pkl

#########################
git 主页的Evaluation权重需要手动点击链接下载
./tools/download_pretrained.sh   # 这个命令行失效了，只能建立pretrain文件夹

cd .. # github项目目录
python setup.py develop

##################################################################
输出
##################################################################
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/spconv/sparse_conv_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/spconv
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/bev_pool/bev_pool_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/bev_pool
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/iou3d/iou3d_cuda.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/iou3d
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/voxel/voxel_layer.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/voxel
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/roiaware_pool3d/roiaware_pool3d_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/roiaware_pool3d
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/ball_query/ball_query_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/ball_query
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/knn/knn_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/knn
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/paconv/assign_score_withk_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/paconv
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/group_points/group_points_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/group_points
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/interpolate/interpolate_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/interpolate
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/furthest_point_sample/furthest_point_sample_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/furthest_point_sample
copying build/lib.linux-x86_64-cpython-38/mmdet3d/ops/gather_points/gather_points_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/gather_points
Creating /home/cine/miniconda3/envs/fusionbev/lib/python3.8/site-packages/mmdet3d.egg-link (link to .)
Removing mmdet3d 1.0.0rc0 from easy-install.pth file
Adding mmdet3d 0.0.0 to easy-install.pth file
detected new path '/home/cine/Documents/GitHub/bevfusion/mmdetection3d'

Installed /home/cine/Documents/GitHub/bevfusion
Processing dependencies for mmdet3d==0.0.0
Finished processing dependencies for mmdet3d==0.0.0

##################################################################

torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox

torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth 

###################
#下载openmpi
# 参考 https://blog.csdn.net/qq_26822029/article/details/107930758
到openmpi官网下载openmpi，这里我使用的是4.0.4版本，也可以使用wget下载：

wget https://download.open-mpi.org/release/open-mpi/v4.0/openmpi-4.0.4.tar.gz

下载完成之后解压：

tar -zxvf openmpi-4.0.4.tar.gz

进入openmpi-4.0.4文件夹之后检查配置文件

cd openmpi-4.0.4
./configure

在漫长的等待之后编译安装

sudo make all install

编译完成之后需要配置一下openmpi的环境变量

vim /etc/profile

# 在这个文件末尾添加如下两行
export PATH=/usr/local/path:$PATH
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib

编辑完成之后需要执行source /etc/profile使其生效

进入到examples文件夹中，执行make编译一下测试代码，如果编译没有报错执行下面的测试语句

make # 先make再执行下面一句 ！！！！！
mpirun -np 4 hello_c

如果正常输出结果则说明安装完成~

```

chg0901, 2023-10-05:
@songhan @zhijian-liu @hopef @K-Shao 
I am sorry for this @ to you, could you please look at this issue and give me some hints? Thank you very much~~

chg0901, 2023-10-05:
I find the solution in here. [link](https://github.com/mit-han-lab/bevfusion/issues/480#issuecomment-1667112241)

closed 2024-05-04 by zhijian-liu

silaopi, 2024-05-11:
Hi, I got the same issue with you. I didn't follow the solution here. Could you please tell me how to solve it? Thank you very much.

> I find the solution in here. [link](https://github.com/mit-han-lab/bevfusion/issues/480#issuecomment-1667112241)