mit-han-lab/bevfusion #518: Training by nuscenes-mini

hsingyu-chou, 2023-10-02:
Hello ,

Sorry , I am a beginner.
I would like to check if I succeed to train the model.
I use google colab pro. The environment configuration is as follows.

GPU Tesla T4
Python 3.10.12 
PyTorch 2.0.1+cu118
CUDA 11.8
cuDNN  8700
mpi4py 3.1.4
Pillow 9.4.0
tqdm 4.66.1
torchpack 0.3.1
Plotnine 0.10.1
Matplotlib 3.5.2
Ninja 1.11.1
Numpy 1.23.5
Numba 0.56.4
Shapely 1.8.5
yapf 0.40.1
flash-attn-1.0.5
MMEngine 0.8.4
MMCV 1.7.1
MMDetection 2.28.1
nuscenes-devkit 1.1.11

I have already copied the 'bevfusion' package and the 'nuscenes-mini' from Google Drive to Colab's disk space. I have also executed the 'setup.py' and 'create.py' commands. However, when I try the 'Tool/Train.py' command, it seems to stop at 'logger.info(f"Model:\n{model}")' without proceeding further, and no errors are thrown.

The system's RAM usage increases, but the GPU RAM (15GB) is only using 0.8GB. I have waited for 18 hours with no results, as shown in the image. Could you confirm if the training is in progress, or if there is something I need to modify?

Command : 
!torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth --run-dir train_results

Picture :
[image]

closed 2023-10-02 by hsingyu-chou

2460707925, 2024-03-07:
I have the same problem，do you solve this ,I check it stop at MMDistributedDataParallel , and i think it maybe multi gpu training problem ,and i train it in one gpu successfully

2460707925, 2024-03-09:
> Hello ,
> 
> Sorry , I am a beginner. I would like to check if I succeed to train the model. I use google colab pro. The environment configuration is as follows.
> 
> GPU Tesla T4 Python 3.10.12 PyTorch 2.0.1+cu118 CUDA 11.8 cuDNN 8700 mpi4py 3.1.4 Pillow 9.4.0 tqdm 4.66.1 torchpack 0.3.1 Plotnine 0.10.1 Matplotlib 3.5.2 Ninja 1.11.1 Numpy 1.23.5 Numba 0.56.4 Shapely 1.8.5 yapf 0.40.1 flash-attn-1.0.5 MMEngine 0.8.4 MMCV 1.7.1 MMDetection 2.28.1 nuscenes-devkit 1.1.11
> 
> I have already copied the 'bevfusion' package and the 'nuscenes-mini' from Google Drive to Colab's disk space. I have also executed the 'setup.py' and 'create.py' commands. However, when I try the 'Tool/Train.py' command, it seems to stop at 'logger.info(f"Model:\n{model}")' without proceeding further, and no errors are thrown.
> 
> The system's RAM usage increases, but the GPU RAM (15GB) is only using 0.8GB. I have waited for 18 hours with no results, as shown in the image. Could you confirm if the training is in progress, or if there is something I need to modify?
> 
> Command : !torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-det.pth --run-dir train_results
> 
> Picture : ![Training_nuscenes-mini](https://private-user-images.githubusercontent.com/146720243/271945611-8e08ea89-9e8f-4743-980a-845eb0413297.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5ODQ2MTIsIm5iZiI6MTcwOTk4NDMxMiwicGF0aCI6Ii8xNDY3MjAyNDMvMjcxOTQ1NjExLThlMDhlYTg5LTllOGYtNDc0My05ODBhLTg0NWViMDQxMzI5Ny5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwOVQxMTM4MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hM2EwM2E3Y2M0N2QzZjNjYmZkODJmMzY1ZmRlOWZhOWZkN2RhY2QzNjAyZGViMWRhYWFmNDc4YWRjNGJkYjIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.oNcwcaE1P3FFrlqq4q5mVOHf2iRbuBm4XSxRxstBcdQ)

I try to transfer transfusion code to solve it ,but failed . Then I set export NCCL_P2P_DISABLE=1 ,it works ,and it is explained in this blog :https://zhuanlan.zhihu.com/p/607203976?utm_id=0