mit-han-lab/bevfusion #519: Visualization error

hsingyu-chou, 2023-10-07:
I try to visualize the results but it occur the error as below.
How could I solve with this issue?

My command :
```
!torchpack dist-run -np 1 python tools/visualize.py train_results/configs.yaml --mode pred --checkpoint train_results/latest.pth --bbox-score 0.5 --out-dir vis_results

!torchpack dist-run -np 1 python tools/visualize.py train_results/configs.yaml --mode gt --checkpoint train_results/latest.pth --bbox-score 0.5 --out-dir vis_results
```

The error :
```
Traceback (most recent call last):
  File "/content/bevfusion/tools/visualize.py", line 166, in <module>
    main()
  File "/content/bevfusion/tools/visualize.py", line 88, in main
    outputs = model(**data)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1501, in _call_impl
    return forward_call(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/parallel/distributed.py", line 1156, in forward
    output = self._run_ddp_forward(*inputs, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/mmcv/parallel/distributed.py", line 165, in _run_ddp_forward
    return module_to_run(*inputs[0], **kwargs[0])  # type: ignore
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1501, in _call_impl
    return forward_call(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/mmcv/runner/fp16_utils.py", line 119, in new_func
    return old_func(*args, **kwargs)
  File "/content/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward
    outputs = self.forward_single(
  File "/usr/local/lib/python3.10/dist-packages/mmcv/runner/fp16_utils.py", line 119, in new_func
    return old_func(*args, **kwargs)
  File "/content/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 301, in forward_single
    feature = self.extract_camera_features(
  File "/content/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1501, in _call_impl
    return forward_call(*args, **kwargs)
  File "/content/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/mmcv/runner/fp16_utils.py", line 208, in new_func
    return old_func(*args, **kwargs)
  File "/content/bevfusion/mmdet3d/models/vtransforms/base.py", line 350, in forward
    x = self.get_cam_feats(img, depth, mats_dict)
  File "/usr/local/lib/python3.10/dist-packages/mmcv/runner/fp16_utils.py", line 208, in new_func
    return old_func(*args, **kwargs)
TypeError: DepthLSSTransform.get_cam_feats() takes 3 positional arguments but 4 were given
```

hsingyu-chou, 2023-10-07:
Temporary solution
/bevfusion/mmdet3d/models/vtransforms/base.py
        x = self.get_cam_feats(img, depth, mats_dict)
→     x = self.get_cam_feats(img, depth)

It could successfully executed but I am not sure if it's correct.
I just skip it.

NuanBaobao, 2023-12-28:
I am also experiencing this error, how did you fix it?

kongwah, 2024-02-06:
Hello, I encountered a similar issue. Any solution?

TimDing-aoi, 2024-03-19:
I got the same error.

kongwah, 2024-03-20:
I resolved it by the following:
-- compare the code between test.py and visualize.py
-- since test.py works, check what is in test.py that is missing in visualize.py 
-- one such missing code snippet relates to wrap_fp16_model 
-- modify visualize.py accordingly to be the same as test.py

bharathsivaram10, 2024-04-23:
This is the same as #532, this should help: [solution](https://github.com/mit-han-lab/bevfusion/issues/532#issuecomment-1886443570)

closed 2024-05-04 by zhijian-liu