mit-han-lab/bevfusion #521: the download_pretrained.sh dosen't work

wangjunquan911, 2023-10-09:
I use colab to download the pretained by using: !sh tools/download_pretrained.sh
Then the result is as follows:

--2023-10-09 09:49:08--  https://bevfusion.mit.edu/files/pretrained_updated/bevfusion-det.pth
Resolving bevfusion.mit.edu (bevfusion.mit.edu)... 18.25.16.171
Connecting to bevfusion.mit.edu (bevfusion.mit.edu)|18.25.16.171|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/bevfusion-det.pth [following]
--2023-10-09 09:49:08--  http://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/bevfusion-det.pth
Resolving hanlab.mit.edu (hanlab.mit.edu)... 44.238.31.106, 44.227.170.101, 44.238.105.202
Connecting to hanlab.mit.edu (hanlab.mit.edu)|44.238.31.106|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/bevfusion-det.pth [following]
--2023-10-09 09:49:08--  https://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/bevfusion-det.pth
Connecting to hanlab.mit.edu (hanlab.mit.edu)|44.238.31.106|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-10-09 09:49:09 ERROR 404: Not Found.

minruixu, 2023-10-09:
I also have this problem!

zhijian-liu, 2023-10-14:
Please use `hanlab18.mit.edu` in place of `hanlab.mit.edu` in the link as a temporary solution.

closed 2023-10-14 by zhijian-liu